openhab/org.openhab.binding.zwave #6: Remove lib folder

kaikreuzer, 2016-07-31:
I wonder why you have mockito and junit at jars in this repo? For compiling & executing the unit tests, it should be alright to simply have Maven dependencies declared in the pom.xml?

cdjackson, 2016-07-31:
I couldn’t work out how to make the tests work from within Eclipse without this. It’s fine under maven, but I also want to be able to run the tests in the IDE…

Is there a way to link the libraries in the IDE without including them in the repo?

kaikreuzer, 2016-07-31:
Hm, all libs are available in the target platform, so you need to get them on your project classpath. The simplest way is to add the required packages as optional dependencies to your MANIFEST.MF. This isn't too nice as it ends up in the jar, but it also does not really harm.

cdjackson, 2016-07-31:
Yes - but that is what I was trying to avoid - it over doubles the size of the JAR (an extra 1.5MB). Doing it this way solved that problem at the expense of adding the files into the repository.

kaikreuzer, 2016-07-31:
Ok then for now. I still hope we will get to bnd some time in future and then we can drop the libs for sure...

closed 2016-07-31 by kaikreuzer