openhab/org.openhab.binding.zwave #9: COMMAND_CLASS_CLIMATE_CONTROL_SCHEDULE command class

cdjackson, 2016-08-02:


marcotglb, 2016-08-29:
Hi Chris.
I've just trying to integrate a POPP Radiator Thermostat.
It is the same of the Danfoss LC-13 but with a richer interface and one of its command classes is COMMAND_CLASS_CLIMATE_CONTROL_SCHEDULE
I'm using the AEOTEC Z-Stick GEN5 as a primary controller.

Right now it appears as an 'Unknown Device' in OH2 'Things' report.
Are you perhaps working on this same device ?

I could do some work myself but it is not clear to me what is the full process to
- update the database (if needed)
- have the new DB working on my installation.

Thanks a lot. Regards.
Marco

maxberger, 2017-03-18:
@marcotglb: Integrating your device with BASIC functionality (e.g. set temperature from the controller) is probably just a matter of a database update. Given that your request is older, this may already have happend.

What would be interesting is actually implementing the climate control schedule, which would allow your thermostat to set the temperature without going back to the controller. I am interested in eventually having this fuctionality, and may be working on it;

@cdjackson: I have a few questions on how this would be implemented properly:
* Climate control schedule needs a UI. Something where the user can actually set the schedule. What would be the right place to put this? What external classes are needed?
* Climate control will also report on the current state; this is currently implemented with a dummy that just logs the value, but eventually that should become a signal. What would be the best way to expose that?
* Climate control needs to keep a counter, which will be increased on every update, so there is the need to store this somewhere. Again: Where / how do I do that?

I am looking for pointers to classes, examples, best practies, etc. I am also looking to see if there is current work on this so that I don't duplicate anything. And last but not least: Should this go on the development branch?

Thanks