openhab/org.openhab.binding.zwave #16: Create MULTI_INSTANCE option to swap source/dest endpoints

cdjackson, 2016-08-08:
It seems some devices (Greenwave) have the source and destination endpoints swapped. The current workaround doesn't work properly and will be reverted.

A suggested workaround is to add an option to the database to specifically swap this for specific devices to ensure that it is controlled to avoid problems with newer devices.

https://github.com/openhab/openhab2-addons/issues/738
https://github.com/openhab/openhab2-addons/pull/742

closed 2016-08-21 by cdjackson

jspuij, 2016-12-20:
@cdjackson config option wont work as only half of the messages are incorrect on the greenwave. Now it swaps all. I really need detection. Why doesn't the detection work on zwave plus nodes?

cdjackson, 2016-12-21:
Because your previous checks made an assumption about endpoint 0. With the latest zwave definition, endpoint 0 is now a valid endpoint where it wasn’t previously.  I forget exactly what you did now, but it caused problems with a number of new nodes that send multi instance data to endpoint 0.

Can you remind me of the logic you want to apply?

jspuij, 2016-12-21:
Yeah, i had another look at the spec indeed, and it seems now that endpoint 0 in the source byte seems to indicate that the Multi instance encapsulation originated from a root command class (and not from an endpoint).

MI encapsulation has a source endpoint byte and a destination endpoint byte:

- source indicates the endpoint on the source node, where the message originated from. Normally between 0 and 127. 
- destination indicates the endpoint on the destination node. This is a number between 0 and 127 normally. (Excluding > 127 for broadcasts, this is not implemented yet in the binding).

The 0 value seems to indicate that the message originated from a root command class. destination 0 seems to indicate that the message is sent to a root command class. (For backwards compatibility all root command classes must also be implemented on endpoint 1 as per spec). It seems my code does not handle source or destination 0, because I've never had a message on my network with source or destination 0.

The controller only receives encapsulated messages that should have a destination value of 1 or 0. The controller has no endpoints (but it might have them in Z-Wave plus, I'm not sure).

What happens with the green-wave device is that because of a bug in the firmware, all unsolicited messages (unsolicited, meaning that it is not a reply to a request from the controller) have the source and destination byte swapped. Replies to messages from the controller are correct.

Since it's illogical for the controller to get a message with a destination != 1, as it has no endpoints, I then assume that source and destination are swapped. I took a general assumption, because most hardware is based on firmware examples from the SDK, so I assumed that more hardware would make this mistake.

You understand now why the option setting is not good enough: Only unsolicited messages are incorrect. You have to detect this.

I realize now that my code would swap bytes if source = 1 and destination != 1 and that this might not be correct:

- If we receive an encapsulation message with destination= 0 and source = 1. This would indicate that the destination is the Z-Wave stick and the originating message was from a command class that is on endpoint 1 of the sending node. This would get swapped, and then fail on the hashmap lookup of endpoints.

- If we receive an encapsulation message with destination > 1 and destination <= 127 and source = 1. This would indicate that the destination is a dynamic endpoint on the Z-Wave stick and the originating message was from a command class that is on endpoint 1 of the sending node. This would get swapped, while it is incorrect. Dynamic endpoints on a bridge controller is not something the binding supports currently, so I don't expect these messages.

i If we receive an encapsulation message with destination > 127 and source = 1. This would be a broadcast message to multiple (dynamic) endpoints on the Z-Wave stick. This would get swapped, while it is incorrect. Same applies as above.

- Further problems with the code as it is now: (not related to my swap)

- source endpoint = 0 is not handled. This would be if the originating command class is on the root of the node and somehow it wants to send to an endpoint on the controller. Not very logical as we don't implement dynamic endpoints on the stick, but if such a packet arrives, the hashmap lookup fails without the swap.

I think my analysis is correct and the case that you mention is where destination= 0 and source = 1. This gets swapped and fails. This could be easily solved by coding something like:

```
if (destinationEndpointId > 1 && originatingEndpointId == 1) {
                logger.debug(
                        "NODE {}: Controller has no endpoints. Probably originating ({}) and destination ({}) endpoints should be swapped.",
                        getNode().getNodeId(), originatingEndpointId, destinationEndpointId);
                // not a full swap. Do not use destinationEndpointId after this line
                // and leave scope intact.
                originatingEndpointId = destinationEndpointId;
            }
```

and it might be best to solve originating endpoint = 0 by either applying 0 = 1 or resolving the command class on the node in the usual way. 

You agree and want me to create a pull request?

cdjackson, 2016-12-21:
I think I would still prefer to make this an option so that we future proof the system. Otherwise we’re implementing quite a nasty hack to fix one device that might bite us in the backside later.

If you can create a PR that wraps this in an option set in the database, then I think it’s fine.

jspuij, 2016-12-21:
I'd say:

1. Log when the option is not set, it's useful info. Might be smart to up the level a bit and include information about where to add the option in the node XML or refer to the online database.
2. Apply workaround if option is set.

Regardless source = 0 should be fixed, because this creates errors.

cdjackson, 2016-12-21:
Ok for me.

I’m not sure what you mean by source=0 should be fixed. Source endpoint can change in multi-instance so it’s not fixed.

jspuij, 2016-12-21:
If the source endpoint in an incoming message is 0, (which would be rare, because this would indicate that the node has encapsulated the message for an endpoint on the controller), the lookup in the hashmap will fail and the message would be discarded...

cdjackson, 2016-12-21:
I don’t want to make these sort of assumptions though. I agree, at the moment it shouldn’t/won’t happen, but I prefer to avoid designing like this and getting bitten in future.

PR #41 (2016-08-21): Add option to use destination endpoint as source
This adds an option to allow the destination endpoint to be used as the source. This seems to be a bug in some devices. It also removes the auto swapping which has problems with some ZWave plus devices that now make use of endpoint 0.

The ```useDestination=true``` option enables this.

https://github.com/openhab/openhab2-addons/pull/742 is reverted with this PR.

Closes #16.
commit 0489ddbd (2016-08-21): Add option to use destination endpoint as source (#41)

This adds an option to allow the destination endpoint to be used as the source. This seems to be a bug in some devices. It also removes the auto swapping which has problems with some ZWave plus devices that now make use of endpoint 0.

The useDestination=true option enables this.

openhab/openhab2-addons#742 is reverted with this PR.

Closes #16.