openhab/org.openhab.binding.zwave #23: Possibility to issue a manual GET Request in Rules

spacemanspiff2007, 2016-08-14:
Hi,

I have some mains powered devices which values I do not read regularly (I don't need them to update that frequent).
However, it would be really nice if there would be the possibility to request values through rules.
So if an updated value is needed, this can be triggered manually.

This can be a function which requires an already existing item, so this should not create too much effort. An optional timeout which waits for the update would be nice too.

Or the function could take a whole configuration string for maximum flexibility. But then the timeout has to be mandatory.

Greetings

cdjackson, 2016-08-14:
It's not possible to pass the sort of information into the binding that you are asking for. Openhab has defined interfaces on how this works. 

It is possible to request a refresh if a value using the refresh command and this likely does most of what you want. this is discussed in the forum so please take a look there.

closed 2016-08-14 by cdjackson

spacemanspiff2007, 2016-08-14:
The ````RefreshType.REFRESH```` will work - thanks.