openhab/org.openhab.binding.zwave #43: Aeotec Minimote failing to be recognized

mrzacharycook, 2016-08-21:
When attempting to add a DSA03202B-ZWUS Aeotec Minimote to Openhab2 via autodiscovery, the following is logged:

2016-08-21 05:02:53.081 [ERROR] [ve.internal.protocol.ZWaveController] - Neither inclusion nor exclusion was active!
2016-08-21 05:02:53.186 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 2: Device could not be resolved to a thingType! 7FFFFFFF:7FFFFFFF:7FFFFFFF::0.0
2016-08-21 05:02:53.411 [INFO ] [smarthome.event.InboxAddedEvent     ] - Discovery Result with UID 'zwave:device:36e36005:node2' has been added.
2016-08-21 05:02:53.415 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zwave:device:36e36005:node2' to inbox.

Attributes from the Thing configuration page are:

zwave_class_basic	CONTROLLER
zwave_class_generic	REMOTE_CONTROLLER
zwave_deviceid	2147483647
zwave_frequent	false
zwave_nodeid	2
zwave_version	0.0
zwave_listening	false
zwave_routing	false
zwave_beaming	true
zwave_class_specific	PORTABLE_REMOTE_CONTROLLER
zwave_manufacturer	2147483647
zwave_devicetype	2147483647

Any ideas why it wouldn't be picking up the deviceid correctly?

cdjackson, 2016-08-21:
Please read the readme for this binding. I know this isn't currently compiled into the docs due to the separate repository but it should be soon. 

Cheers
Chris

cdjackson, 2016-08-21:
Just to be clear - this is the main bit I was referring to -:

Unknown Devices

If the device is listed as Unknown, then the device has not been fully discovered by the binding. There are a few possible reasons for this -:

The device is not in the database. If the device attributes show that this device has a valid manufacturer ID, device ID and type, then this is likely the case. Even if the device is in the database, some manufacturers use multiple sets of references for different regions or versions, and your device references may not be in the database. In either case, the database must be updated and you should raise an issue to get this addressed.
The device initialisation is not complete. Once the device is included into the network, the binding must interrogate it to find out what type of device it is. One part of this process is to get the manufacturer information required to identify the device, and until this is done, the device will remain unknown. For mains powered devices, this will occur quickly, however for battery devices the device must be woken up a number of times to allow the discovery phase to complete. This must be performed with the device close to the controller.

closed 2016-08-21 by cdjackson