openhab/org.openhab.binding.zwave #50: Missing requirement at startup (com.google.gson v2.5.0)

mhilbush, 2016-08-22:
Last night's build (#30) is complaining about a missing requirement at startup: com.google.gson v2.5.0

```
2016-08-22 11:30:15.509 [ERROR] [core.karaf.internal.FeatureInstaller] - Failed installing 'openhab-binding-zwave': Unable to resolve root: missing requirement [root] osgi.identity; osgi.identity=openhab-binding-zwave; type=karaf.feature; version="[2.0.0.SNAPSHOT,2.0.0.SNAPSHOT]"; filter:="(&(osgi.identity=openhab-binding-zwave)(type=karaf.feature)(version>=2.0.0.SNAPSHOT)(version<=2.0.0.SNAPSHOT))" [caused by: Unable to resolve openhab-binding-zwave/2.0.0.SNAPSHOT: missing requirement [openhab-binding-zwave/2.0.0.SNAPSHOT] osgi.identity; osgi.identity=org.openhab.binding.zwave; type=osgi.bundle; version="[2.0.0.201608221458,2.0.0.201608221458]"; resolution:=mandatory [caused by: Unable to resolve org.openhab.binding.zwave/2.0.0.201608221458: missing requirement [org.openhab.binding.zwave/2.0.0.201608221458] osgi.wiring.package; filter:="(&(osgi.wiring.package=com.google.gson)(version>=2.5.0))"]]
```

The karaf console shows only 2.3.1 is installed.

```
openhab> list -s | grep google
 12 | Active   |  80 | 2.3.1                 | com.google.gson
 13 | Active   |  80 | 18.0.0                | com.google.guava
 14 | Active   |  80 | 3.0.0.v201312141243   | com.google.inject
```

cdjackson, 2016-08-22:
I don’t think it’s last nights build, but todays - I only merged this 2 hours ago…

I’ll remove the version and hopefully that will fix it.

mhilbush, 2016-08-22:
Ah, yeah.  I see there was a new version built on cloudbees about an hour ago.  I'm running the online version of OH so it must've tried to pick that up!

cdjackson, 2016-08-22:
Yep. I’ve just put in a PR - once the tests have built I’ll merge and then CB should rebuild…

cdjackson, 2016-08-22:
Looks like it’s built… See if that solves the problem.

mhilbush, 2016-08-22:
That fixed it.  Thanks!

closed 2016-08-22 by mhilbush