openhab/org.openhab.binding.zwave #54: NPE during node initialization

mhilbush, 2016-08-25:
I saw this ArrayOutOfBounds exception during initialization of a node.  Further detail in attached file.  I don't recall seeing this before, so I'm wondering if the message was corrupted.

```
2016-08-24 14:44:31.788 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Node advancer: loop - DELETE_ROUTES try 1: stageAdvanced(false)
2016-08-24 14:44:31.788 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Node advancer - advancing to RETURN_ROUTES
2016-08-24 14:44:31.789 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveInitializationStateEvent
2016-08-24 14:44:31.789 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 20: Got an event from Z-Wave network: ZWaveInitializationStateEvent
2016-08-24 14:44:31.790 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Node advancer: loop - RETURN_ROUTES try 0: stageAdvanced(true)
2016-08-24 14:44:31.790 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 20: Generate return routes list
2016-08-24 14:44:31.790 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Adding return route to 1
2016-08-24 14:44:31.790 [DEBUG] [essage.AssignReturnRouteMessageClass] - NODE 20: Assigning return route to node 1
2016-08-24 14:44:31.790 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Node advancer - queued packet. Queue length is 1
2016-08-24 14:44:31.791 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.ArrayIndexOutOfBoundsException: 3
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSecurityCommandClass.isSecurityNonceReportMessage(ZWaveSecurityCommandClass.java:1067)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.SerialMessage$SerialMessageComparator.compare(SerialMessage.java:692)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.SerialMessage$SerialMessageComparator.compare(SerialMessage.java:1)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at java.util.concurrent.PriorityBlockingQueue.siftUpUsingComparator(PriorityBlockingQueue.java:374)[:1.8.0_91]
	at java.util.concurrent.PriorityBlockingQueue.offer(PriorityBlockingQueue.java:491)[:1.8.0_91]
	at java.util.concurrent.PriorityBlockingQueue.add(PriorityBlockingQueue.java:463)[:1.8.0_91]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.enqueue(ZWaveController.java:520)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.sendMessage(ZWaveNodeInitStageAdvancer.java:260)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.addToQueue(ZWaveNodeInitStageAdvancer.java:1152)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.advanceNodeStage(ZWaveNodeInitStageAdvancer.java:1064)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.handleNodeQueue(ZWaveNodeInitStageAdvancer.java:230)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.ZWaveIncomingEvent(ZWaveNodeInitStageAdvancer.java:1288)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:541)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:244)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:208)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:202)[175:org.openhab.binding.zwave:2.0.0.201608232047]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1298)[175:org.openhab.binding.zwave:2.0.0.201608232047]
2016-08-24 14:44:36.523 [ERROR] [ocol.ZWaveController$ZWaveSendThread] - NODE 255: Timeout while sending message. Requeueing - 2 attempts left!
2016-08-24 14:44:36.523 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 41. Queue={}
2016-08-24 14:44:36.525 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 40
2016-08-24 14:44:36.525 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 05 00 47 14 01 A8 
2016-08-24 14:44:36.526 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 05 00 47 14 01 A8 
2016-08-24 14:44:36.538 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 47 01 BC 
2016-08-24 14:44:36.539 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-08-24 14:44:36.539 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 47 01 BC 
2016-08-24 14:44:36.540 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 47 01 BC 
2016-08-24 14:44:36.540 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=DeleteReturnRoute[0x47], type=Response[0x01], priority=High, dest=255, callback=0, payload=01 
2016-08-24 14:44:36.540 [DEBUG] [essage.DeleteReturnRouteMessageClass] - NODE 20: Got DeleteReturnRoute response.
2016-08-24 14:44:36.541 [DEBUG] [essage.DeleteReturnRouteMessageClass] - NODE 20: DeleteReturnRoute command in progress.
2016-08-24 14:44:36.788 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 05 00 47 01 00 BC 
2016-08-24 14:44:36.788 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 20: Stage RETURN_ROUTES. Initialisation retry timer triggered. Increased to 10000
```
It looks like the code is testing for payload length < 3, then accessing bytes 3 and 4 of the payload?
```
if (serialMessage.getMessagePayload() == null || serialMessage.getMessagePayload().length < 3) {
    return false;
}
byte[] payloadBytes = serialMessage.getMessagePayload();
boolean result = (payloadBytes[2] == ((byte) CommandClass.SECURITY.getKey())
        && payloadBytes[3] == SECURITY_NONCE_REPORT);
```

Detailed debug log.
[aiob-exception.txt](https://github.com/openhab/org.openhab.binding.zwave/files/436765/aiob-exception.txt)

mhilbush, 2016-09-15:
I expect this commit will prevent this from occurring in current builds?

https://github.com/openhab/org.openhab.binding.zwave/commit/c2343cb6c4ca467e43f1f13aaa5133532f2d26d3

However, the array index issue still exists so we should leave this issue open?

cdjackson, 2016-11-28:
Going to close this - the security class has been completely rewritten so I doubt this problem will exist (there will be others instead :) ).

closed 2016-11-28 by cdjackson