openhab/org.openhab.binding.zwave #55: Feature request: Periodically check for device ONLINE/OFFLINE status based on wakeup interval settings

IOOOTAlan, 2016-08-25:
Greetings,

as discussed [on the forum](https://community.openhab.org/t/z-wave-battery-devices-issues-in-general/10022/52), it would be desirable for the ZWave binding to update the Thing status for ZWave devices by using the wakeup interval and last wakeup information already available.
This is especially true in battery-powered devices, as they currently can physically disappear from the network (e.g. battery removed or completely discharged) while keeping their "ONLINE" status indefinitely.
I guess this enhancement could be implemented by checking periodically, for every ZWave Thing currently ONLINE, the condition
`NOW - (zwave_wakeup_time + wakeup_interval) > SET_OFFLINE_TIME_THRESHOLD`
where
 - `NOW` would be the current time
 - `zwave_wakeup_time` is the Thing property reporting the last time a wakeup signal from the device was observed
 - `wakeup_interval` is how often the Thing performs a wakeup (in the Thing configuration settings)
 - `SET_OFFLINE_TIME_THRESHOLD` would be a tolerance threshold (ideally close to 0) after which the Thing is to be considered definitely offline.
On a side note, we observed that mains-powered devices are not set OFFLINE when removed from the power source unless we try to update their status: is this expected behavior?

cdjackson, 2016-08-25:
Mains powered devices will be placed OFFLINE when they are polled or otherwise accessed.

IOOOTAlan, 2016-09-19:
Great! Will test it soon :+1:

IOOOTAlan, 2016-09-20:
I think there's something wrong: one of my devices seems to not come back online even after causing a wakeup event:
[image]
The screenshot was taken just after sending the wakeup (pushing the button on the device) precisely at `2016-09-20 08:14:19,517`.
Could this be a side effect of PR #100?

cdjackson, 2016-09-20:
Yes - you might be right. Since the device is not considered DEAD by the binding, it is not restored - I’ll take a look at this.

cdjackson, 2016-09-21:
I will merge an updated version of this tonight that I hope will work better - please let me know how it goes.

closed 2016-09-21 by cdjackson

IOOOTAlan, 2016-09-22:
I just finished testing it and it seems to work flawlessly :+1: 
The node was successfully set as "Online" after reinserting the batteries and forcing a wake up event.

cdjackson, 2016-09-22:
Great - thanks.

PR #100 (2016-09-16): Add check to make battery node DEAD if no message in twice the wakeup period
closes #55
PR #119 (2016-09-21): Set battery node to DEAD if no response within 2x wakeup period
fixes #55
commit 885140e7 (2016-09-21): Set battery node to DEAD if no response within 2x wakeup period (#119)

fixes #55