openhab/org.openhab.binding.zwave #58: [transaction] Repeated CONFIG_GET message sent every second or so during init

mhilbush, 2016-08-29:
PR build 53

My network is:
Node 1 - Aeon Z-Stick Gen 5
Node 5 - Fibaro FGMS-001
Node 7 - Aeon ZW100
Node 8 - Linear PD300Z Dimmer

I woke up node 7 multiple times to initialize it.  The first wakeup was around 7:30:13.  I observed CONFIG_GET messages being sent about every second or so.  The messages stopped once the device was fully initialized.

Debug log is attached.  I log zwave to a separate file, so there are only zwave messages in the file.
[zwave-multiple-config-get.txt](https://github.com/openhab/org.openhab.binding.zwave/files/442345/zwave-multiple-config-get.txt)

cdjackson, 2016-08-29:
While there might be an issue with managing the "no more information” message while a transaction is outstanding, I think the problem with the config messages might be the database. It looks like the devie isn’t responding to the request - it asks it, but never sends back the information. This is normally a sign of the parameter not being supported.

What version of the device do you have (there are at least 3, and some do, and some don’t have parameter 8).

mhilbush, 2016-08-29:
Node7
Manufacturer 0086
Type / ID 0102:0064
Firmware Version 1.7

I believe you are correct.  The 1.7 spec doesn't list parameter 8.

I can do a full comparison of the 1.7 spec against the DB, just in case parameter 8 is not the only one in error. Ok?

mhilbush, 2016-08-29:
Well, hang on there.  There are two 1.7 specs attached to the database entry.  One lists parameter 8 and the other doesn't.

cdjackson, 2016-08-29:
You mean two Aeon specs for 1.7?

Great! ;)

mhilbush, 2016-08-29:
Exactly!  Welcome to ZWave.  

I just printed both specs and I'll do a comparison.

mhilbush, 2016-08-30:
The one ZW100 v1.7 spec is dated 3/16/2016.  The other ZW100 v1.7 spec is dated 5/11/2016.  

The change history in the 5/11 spec says it adds "more details for config parameters."  In fact, the 5/11 spec adds a bunch of new config parameters.  Not only did it add parameter 8, it also added parameters 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, and 64.

The database entry for ZW100 v1.7 matches the 3/16 spec with the exception of parameter 8.

I say we remove parameter 8 from the database for the ZW100 v1.7.  If you agree, I'll make the change.

mhilbush, 2016-08-30:
Scratch that.  Since it's a delete, I think you need to make the change.

cdjackson, 2016-08-30:
> I say we remove parameter 8 from the database for the ZW100 v1.7. If you agree, I'll make the change.

Sounds good - thanks.

cdjackson, 2016-08-30:
I’ve deleted parameter 8 from the 1.7 version. I’ll try and update the binging tonight and will also push the change to the transaction branch...

mhilbush, 2016-09-01:
Confirmed that the deletion of parameter 8 resolved the repeated CONFIG_GET calls.

closed 2016-09-01 by mhilbush

cdjackson, 2016-09-01:
Great - thanks.

So I guess the transaction code is working ok for you? I have one issue I need to iron out before I can merge this anyway, so any further comments appreciated… (this issue might take a week or two)

mhilbush, 2016-09-01:
Yes, running well.  I did open another issue with some questions on other behavior I was seeing.

cdjackson, 2016-09-01:
Thanks - I spotted that after I’d fired off the email. I’ll take a look tonight.