openhab/org.openhab.binding.zwave #64: [transaction] Multiple calls to get temp, humidity, and version

mhilbush, 2016-09-01:
Not sure if these are problems or not, but I'm going to open the issue so that you can have a look.

PR build 61

My network is:
Node 1 - Aeon Z-Stick Gen 5
Node 5 - Fibaro FGMS-001
Node 7 - Aeon ZW100
Node 8 - Linear PD300Z Dimmer
Node 9 - Linear PS15Z Appliance Module (just added)

Node 7 at 7:03:12 in the log
- repeated requests for temperature.  I was waking the device up, so it's possible this was the cause?
- log viewer is labeling this as SENSOR_MULTILEVEL_GET **DIRECTION**. Is this a viewer problem?

Node 5 at 7:03:52 in the log
- repeated request for humidity.  Again, I was waking up the device, so was this the cause?

Node 9 at 7:10:57 in the log
- There are three calls for MANUFACTURER_SPECIFIC_GET.  Is that expected behavior?

Node 9 at 7:11:04 in the log
- There are multiple calls for VERSION_COMMAND_CLASS_GET NO_OPERATION.  Is this a DB issue for this device?

[zwave-2016-09-01.txt](https://github.com/openhab/org.openhab.binding.zwave/files/449520/zwave-2016-09-01.txt)

cdjackson, 2016-09-01:
It looks like sensor_multilevel is sending incorrect commands in this branch - not sure why, but I've added a test and it does detect the problem. That probably covers the first 2 issues (or at least part of them) - the reason it repeats is because the binding asks for a certain sensor, but the device returns a different one - so it asks again...

I've not looked at the other 2 issues yet - probably one for tomorrow night...

cdjackson, 2016-09-01:
I've just pushed an update that will hopefully fix the multilevel sensor issue.

mhilbush, 2016-09-01:
Great. I just pulled build 62, and will test it tonight or tomorrow morning.

mhilbush, 2016-09-02:
The sensor multilevel reporting is looking much better.  Thanks.

cdjackson, 2016-09-02:
I think I’ve now fixed the multiple requests for the MANUFACTURER information - this will get pushed tonight hopefully.

mhilbush, 2016-09-02:
Ok. I should be able to pull down the new build tonight, but I might not be able to test it until Sunday.

cdjackson, 2016-09-02:
No probs - I’ve just pushed the latest update. I’ll try and look over the no-op issue as well before Sunday...

cdjackson, 2016-09-02:
I also found a bug in no-op processing, so that might also be fixed now (although not 100% sure).

mhilbush, 2016-09-02:
Ok.  I see there's a build 65 on CB.  Does that one have the no-op fix?

cdjackson, 2016-09-02:
Yes - looks like 65 should be the one...

mhilbush, 2016-09-02:
Ok.  I downloaded 65 and have it installed.  I'm VPN'ed into home, so I can't wake up any of the battery devices.  I'll just have to wait for them to wake up on their own.  :-(

mhilbush, 2016-09-03:
For Node 9, I'm still seeing repetitive version_command_class_get no_op requests starting at 16:53.46, 16:54:25, 16:56:21, etc. in the logs.  Again, I'm not sure if this is a device issue, database issue, or binding issue.

For nodes 5 and 7, I'm seeing frequent and repetitive get requests being sent when it appears the devices are asleep, but perhaps I'm not interpreting that correctly.  Are these possibly poll requests that are supposed to be queued and sent when the device wakes up?

Other than the above two questions, this version is looking pretty solid.

cdjackson, 2016-09-03:
Can you send an updated log?

I doubt the no-op issue is database or device. I know that there’s a problem with this during startup - not in this branch specifically, but in the binding somewhere.

I’d be interested in the sleeping thing - maybe this transaction manager bypasses the wakeup class queue - I see another test coming for this...

mhilbush, 2016-09-03:
Yeah, as soon as I hit comment I realized I forgot to attach the log.  :-o  Comment above was edited to include the log.

cdjackson, 2016-09-03:
NO_OP problem should now be fixed. I’d fixed this in the master branch a while back, but it must have been after the transaction branch was split and it didn’t merge for some reason...

mhilbush, 2016-09-03:
Confirming no-op is fixed in build 71.

cdjackson, 2016-09-16:
@mhilbush do you think we can close this or is something still missing? I'm tempted to close and open another issue if we find something as I've lost track of what is still open in this.

mhilbush, 2016-09-16:
The only thing I'm not sure about was that the binding was sending transactions to battery devices while they were asleep. If this is resolved, or will be resolved in the rework of the queuing, then this can be closed.

cdjackson, 2016-09-16:
Yep - good point. I’m not sure if I fixed that, but it is/was an issue. I’ll close this and make a specific issue…

closed 2016-09-16 by cdjackson