openhab/org.openhab.binding.zwave #81: [transaction] Why would some messages be going to wrapper.log?

mhilbush, 2016-09-05:
Would there be any reason why some zwave log messages would be going to wrapper.log (and the karaf console)?  Here's a snippet of what's showing up in wrapper.log.

```
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transaction 0 is last transaction
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance ST: WAIT_RESPONSE
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance TX: Message[43]: class=GetRoutingInfo[0x80], type=Request[0x00], dest=255, callback=0, payload=09 00 00 03 
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance WT: null
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance RX: Message[47]: class=GetRoutingInfo[0x80], type=Response[0x01], dest=255, callback=0, payload=00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transaction 0 advanced to DONE
INFO   | jvm 1    | 2016/09/05 09:06:03 | STOP transaction timer
INFO   | jvm 1    | 2016/09/05 09:06:03 | handleTransactionComplete DONE 0
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transaction 0 completed
INFO   | jvm 1    | 2016/09/05 09:06:03 | Sending message Message[46]: class=GetRoutingInfo[0x80], type=Request[0x00], dest=255, callback=0, payload=07 00 00 03 
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transactions outstanding: 1
INFO   | jvm 1    | 2016/09/05 09:06:03 | Timer type WAIT_RESPONSE
INFO   | jvm 1    | 2016/09/05 09:06:03 | STOP transaction timer
INFO   | jvm 1    | 2016/09/05 09:06:03 | Start transaction timer to 500     1473080764209
INFO   | jvm 1    | 2016/09/05 09:06:03 | Received msg Message[50]: class=GetRoutingInfo[0x80], type=Response[0x01], dest=255, callback=0, payload=00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO   | jvm 1    | 2016/09/05 09:06:03 | lastTransaction = WAIT_RESPONSE: callback: 0
INFO   | jvm 1    | 2016/09/05 09:06:03 | Checking outstanding transactions: 1
INFO   | jvm 1    | 2016/09/05 09:06:03 | Last transaction: WAIT_RESPONSE: callback: 0
INFO   | jvm 1    | 2016/09/05 09:06:03 | Message is RESponse GetRoutingInfo   GetRoutingInfo
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transaction 0 is last transaction
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance ST: WAIT_RESPONSE
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance TX: Message[46]: class=GetRoutingInfo[0x80], type=Request[0x00], dest=255, callback=0, payload=07 00 00 03 
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance WT: null
INFO   | jvm 1    | 2016/09/05 09:06:03 | TransactionAdvance RX: Message[50]: class=GetRoutingInfo[0x80], type=Response[0x01], dest=255, callback=0, payload=00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
INFO   | jvm 1    | 2016/09/05 09:06:03 | Transaction 0 advanced to DONE
INFO   | jvm 1    | 2016/09/05 09:06:03 | STOP transaction timer
INFO   | jvm 1    | 2016/09/05 09:06:03 | handleTransactionComplete DONE 0
```

I do have my logging config tweaked to send zwave binding messages to it's own log file.  I have that config on three separate systems, yet the only system where some messages are going to wrapper.log is the system with the binding with the new transactional code.  And, I don't see anything that's different between the three configs.  They all have this in `runtime/karaf/etc/org.ops4j.pax.logging.cfg`

```
# Logger for zwave binding
log4j.logger.org.openhab.binding.zwave = DEBUG, zwave

# Dedicated file appender for zwave binding - zwave.log
log4j.appender.zwave=org.apache.log4j.RollingFileAppender
log4j.appender.zwave.layout=org.apache.log4j.PatternLayout
log4j.appender.zwave.layout.ConversionPattern=%d{yyyy-MM-dd HH:mm:ss.SSS} [%-5.5p] [%-36.36c] - %m%n
log4j.appender.zwave.file=${openhab.logdir}/zwave.log
log4j.appender.zwave.append=true
log4j.appender.zwave.maxFileSize=10MB
log4j.appender.zwave.maxBackupIndex=10
log4j.additivity.org.openhab.binding.zwave = false
```

Of course, the installation of the binding is different.  In this case, it is installed manually in addons.  In the other two cases, it is installed automatically by placing zwave in conf/services/addons.cfg.

In the meantime, I'll revert my logging config to the original config that came with the distro to see if that makes a difference.

mhilbush, 2016-09-05:
If I revert `runtime/karaf/etc/org.ops4j.pax.logging.cfg` to the one that came with the distro, I still see zwave messages going to wrapper.log.

cdjackson, 2016-09-05:
I don’t know why this would happen. I don’t see how it can be anything to do with the binding as it has no control over the logging (AFAIK anyway).

mhilbush, 2016-09-05:
I'm at a loss as well.  I agree that it shouldn't have anything to do with the binding.

I'm going to try a couple other things to see if the behavior stays or goes away.
- install non-PR build into addons directory
- install non-PR build using PaperUI

mhilbush, 2016-09-05:
I don't know what to make of this.

- When I install the non-PR build into addons, it logs correctly.

- When I install the PR build into addons, it logs some messages to wrapper.log and the karaf console.

In both cases above, I'm starting with a clean install.

If you're not seeing this behavior, I'm going to close the issue.

closed 2016-09-05 by mhilbush

pizerg, 2016-09-07:
@mhilbush I think you're getting all "System.out" redirected to your wrapper.log as the transaction PR code has several messages printed via "System.out" instead of/besides logger.xxxx.

cdjackson, 2016-09-07:
Ah - yes, that could be it. I am using system calls in the transaction code at the moment as it helps with debugging in the test environment. They will be removed before merging.

mhilbush, 2016-09-07:
That explains it.  Thanks @pizerg