openhab/org.openhab.binding.zwave #82: [transaction] ZWaveThermostatSetPointCommandClass setMessage ArrayOutOfBounds

pizerg, 2016-09-06:
setMessage method crashes due to out of bounds exception when executing the following code:

`byte[] payload = ArrayUtils.addAll(new byte[] { (byte) setpointType.getKey() }, encodedValue);
payload[5] += (byte) (scale << 3);`

Tried replacing by master's payload and worked properly, this is the payload from master branch:

`byte[] payload = ArrayUtils.addAll(new byte[] { (byte) this.getNode().getNodeId(), (byte) (3 + encodedValue.length), (byte) getCommandClass().getKey(), THERMOSTAT_SETPOINT_SET, (byte) setpointType.getKey() }, encodedValue);`

Maybe changing payload[5] index would fix it as well

mhilbush, 2016-09-06:
@pizerg  So you also are testing the new transaction code?  Would you mind checking to see if you have any zwave debug messages going to wrapper.log?  I was seeing this with only the PR builds of the binding.  I closed the issue because I could find no plausible explanation for why it was happening, and assumed it was just something about my config?  Thanks.

Issue is here.
https://github.com/openhab/org.openhab.binding.zwave/issues/81

cdjackson, 2016-09-15:
I'm just looking at this, and I'm surprised your fix of copying the master worked as this will create a totally invalid command with the new system as it will add the node id and a bunch of other data that will then be added again within the transaction. I suspect it's not really working, but it won't crash so you don't notice.

I'll push a fix shortly - please let me know if it helps.

pizerg, 2016-09-16:
Tested with the new code and works properly on a Remotec IR and a Danfoss valve.
As you say the message created was invalid so last time maybe I did a quick test which just seemed to work but actually did not.

Thanks for the fix!

commit c34e8d67 (2017-02-01): Transaction refactoring (#269)

closes #82
fixes #102
closes https://github.com/openhab/org.openhab.binding.zwave/issues/264
fixes #267
closed 2016-09-16 by pizerg