openhab/org.openhab.binding.zwave #88: Fibaro System FGS223 - unknown device

itrahk, 2016-09-11:
Hi, 
I'm new to openhab so forgive me if I ask obvious questions. 

I've just connected Fibaro Double switch (FGS223) but it is discovered as Unknown Device - Z-Wave Node 2 (010F:0203:1000:3.2). I checked device database, and the device is there (http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/416). Should I do anything else to somehow refresh device database or can import product file manually (it is an export option on database website) to my OH2?

I'm using ZWave Binding version 2.0.0.201609062002

cdjackson, 2016-09-11:
This device has only just been added to the database, but it’s not in the current version. At the moment, builds are disabled, but once they are enabled again the next snapshot should include this device.

itrahk, 2016-09-11:
Is it possible to add device file exported from the database website manually to oh2?

cdjackson, 2016-09-11:
No - this isn’t possible.

The build was just updated, so please try the current version.

itrahk, 2016-09-11:
Now (binding version 2.0.0.201609111700) device is recognized properly. Thanks for help.

closed 2016-09-11 by itrahk