openhab/org.openhab.binding.zwave #91: Display whether Z-Wave Pairing is/was Secure or Non-Secure

jyoukers, 2016-09-13:
It would be nice to see this information.

shorty707, 2016-10-01:
+1 this

Alessandro65, 2016-11-13:
+1 for me too

cdjackson, 2016-11-13:
This is being add as security is implemented - until then it’s pointless as security doesn’t work.

cdjackson, 2016-11-24:
Implemented in security development branch

cdjackson, 2017-02-01:
This is implemented in the development branch so closing.

closed 2017-02-01 by cdjackson

pizerg, 2017-02-01:
Hi @cdjackson ,

Is this related to the field "security" on the serialized xmls ? I found that value not to be reliable as (most) security devices I've tested do not report that bit properly so I ended up adding an extra field that actually reflects when the device has been securely included.

cdjackson, 2017-02-01:
No - it's not related to the security flag in the XML - I agree this is unreliable for most devices and also would just indicate if the node supports security - not if it has actually been securely included in the network. 

The implementation adds a property in ESH that is based on the list of secure command classes. ie if there are secure command classes listed for the node, then we are included securely.

https://github.com/cdjackson/org.openhab.binding.zwave/blob/27bf701347ff2a63fa0b4cd96a2f59d83f62c6a1/src/main/java/org/openhab/binding/zwave/internal/protocol/ZWaveNode.java#L816

pizerg, 2017-02-01:
Ok, thanks for the clarification!