openhab/org.openhab.binding.zwave #101: Linked binary switch not updated

thopiekar, 2016-09-16:
Here is my setup in short:

Node 20: Aeotec ZW100 MultiSensor 6
Node 24: Aeotec DSC27 Micro Illuminator G2

I took HABmin (don't know where that problem here goes to) and linked the binary_switch (motion sensor) to the dimmers (Node 24) switch_dimmer. So far everything works great when looking at the UI.
When going though the room the binary_switch of the sensors node turn on and the dimmers switch_dimmer turns to 100%.
BUT: The light does not turn on. 

Added an entry to my sitemap like this:
```
            Text label="Node 24: DSC27 Micro Illuminator G2" {
                Switch item=zwave_device_f9088262_node24_switch_dimmer
            }
```
And that works sort of.. sometimes jumping around when clicking on it.

(Offtopic: Where are those links saved? Are they kept after updating OH2?)

Regards

cdjackson, 2016-09-16:
Not sure I understand. Do you mean that when you press the switch in the room, the light doesn’t work, but the UI does show the state changed?

Links are stored in the database in OH2.

thopiekar, 2016-09-16:
Exactly. But the switch here is a motion sensor, see http://aeotec.com/z-wave-sensor
Whenever it detects a movement it turns to `ON`.

thopiekar, 2016-09-16:
Oh, you can take the same link, like used in #17, to see the logs: https://www.myqnapcloud.com/smartshare/69237gkn4l6p7008uv2z1x3c_6j9o21i :wink:

cdjackson, 2016-09-16:
Ok so I'm still not sure I understand. It doesn't sound like it's anything to do with the binding if the binding can control the light ok. 

If you're trying to get a motion detector to turn on a light then you need to use associations but again this isn't anything to do with the binding.

thopiekar, 2016-09-16:
Ok, so which OH repository should be the best to report that?

thopiekar, 2016-09-16:
Reporting it on openhab/openhab now...

cdjackson, 2016-09-16:
If the problem is associations, then you need to configure your system - it’s not a bug that you can report and expect to get fixed. You need to configure associations that link the motion sensor to the light.

thopiekar, 2016-09-16:
Like that?:

[image]

cdjackson, 2016-09-16:
> Like that?

I'm not sure what the image is meant to show, but it's nothing related to associations.

thopiekar, 2016-09-16:
Doesn't it work like that? I go to the sensor, choose "channels", go to "Binary Sensor" and add a link there to my dimmer, here "zwave_device_f9088262_node19_switch_dimmer".
Or did I misunderstand something? :confused:

thopiekar, 2016-09-16:
(the second dimmer in the screenshot is another, which I just added for testing)

cdjackson, 2016-09-16:
I’m not 100% sure I understand what you want to do still, but if you want to have the motion sensor directly turn on a light, then it doesn’t work like you said. To do this, you need to configure an association. Normally in the device there will be an associations configuration section - you will need to configure this to link to the other device.

If I do understand what you want, then I’d really suggest closing this and raising the question on the forum where we should be addressing general support issues.

thopiekar, 2016-09-16:
Tried this before and it also didn't work..
Will play around this..

closed 2016-09-16 by thopiekar

cdjackson, 2016-09-16:
If you can’t get it working, then open something on the forum, but please try and be really clear what you’re trying to do ;)

thopiekar, 2016-09-16:
Ok, thanks :smiley: