openhab/org.openhab.binding.zwave #110: Sending custom notification types

thopiekar, 2016-09-18:
Now as the siren (DLink Z510) is working great, the only question which has left is:
Is there a way to play different sounds?

The manual says that I'll have to send different "notification types", see:
http://download.telldus.se/Manuals/Z-Wave/D-Link/DCH-Z510.pdf - Page 3: "Play Sound"

If I'm not wrong there is support now in the Z-Wave binding (compared with the past) to send notifications (as the siren also responds in the NOTIFICATION_REPORT mode).

Regards :smiley:

cdjackson, 2016-09-18:
The binding can’t send notifications - normally other devices send them when there’s an alarm of some description. It could be added, but there’s never been a need in the past so it’s not implemented.

thopiekar, 2016-09-18:
Ok, do you have any ideas how to implement this? Is it a difficult task? Would it take much time?
(Just asking without looking on priorities on other problem/features...)

cdjackson, 2016-09-18:
Yes - it can be implemented. It’s not loads of work (a few hours probably) and I don’t expect it to be difficult. It’s just priorities...

cdjackson, 2016-10-08:
I believe that this is working (as stated on the forum) so closing.

closed 2016-10-08 by cdjackson