openhab/org.openhab.binding.zwave #114: Aeon Labs Smart Strip DSC11-ZWUS - OH2

brianllong, 2016-09-19:
Not sure if this is the proper place but

Please update openhab2 zwave database  to add endpoints 5 and 6

thanks

cdjackson, 2016-09-19:
These endpoints already exist as far as I can tell.

brianllong, 2016-09-19:
i am have troubles getting the last two outlets to show me wattage draw

https://community.openhab.org/t/aeon-labs-smart-strip-dsc11-zwus/3196/13

cdjackson, 2016-09-19:
What version of the binding are you running?

brianllong, 2016-09-19:
2.0 snapshot , i believe it came with openhab2 b4

cdjackson, 2016-09-19:
Ok, if you are using b4 then it’s too old and you need to upgrade.

brianllong, 2016-09-19:
can you point me in the right direction on updating the binding?

cdjackson, 2016-09-19:
It would be easier to upgrade the complete runtime to the latest snapshot rather than using the beta version. You can download the snapshot distribution from openhab.ci.cloudbees.com

brianllong, 2016-09-19:
i wasnt aware openhab 2 was out of beta

cdjackson, 2016-09-19:
Ok, but that’s what you said you were using ;) It’s been on beta testing for nearly a year (with the current version being the 4th beta - hence b4).

brianllong, 2016-09-21:
not sure what going on but i completely loaded a fresh openhab2 with jar file from openhab.ci.cloudbees.com and got the same results missing Wattage for 5 and 6 

hopefully this will fix itself in future updates

thank you again for looking into this

xsnrg, 2016-09-21:
It has been some time, but I have two of these devices and there is something with them outside of openhab where they do not work quite right.  These are first gen, so they may have updated them, but the wattage report endpoints don't line up numerically with the switch endpoints.  That is, when I was testing, switch 1 was endpoint 3 for power reporting.  Once it got to the 5th port, endpoint 7 doesn't exist, so two of the ports don't report power.  I asked Aeon about this, but didn't get much of a reply.  This was about 18 months or more ago.

cdjackson, 2016-09-21:
> On 21 Sep 2016, at 14:18, brianllong <notifications@github.com> wrote:
> 
> hopefully this will fix itself in future updates
> 
> 

Unfortunately, in my experience, things rarely fix themselves ;)

Please can you provide a debug log of the startup with no XML, or at least provide the XML so I can see what the device is reported as.  The XML is in the /userdata/zwave folder.

cdjackson, 2016-11-03:
Closing due to lack of activity. Please re-open if there is further information

closed 2016-11-03 by cdjackson

velaar, 2018-02-18:
Hi @cdjackson, would you mind if I reopen it? Have the same issue with only 4 meters out of 6 working. Switches work fine. Also on DSC11 only sockets 1-4 are switchable so 5,6 switch channels do not seem to be doing anything. Node11.xml is [here.](https://pastebin.com/XQRAvbBx)