openhab/org.openhab.binding.zwave #126: Node persistence files need to allow multiple controllers

cdjackson, 2016-09-24:
Either add separate folder per controller, or add homeid to filename

bodiroga, 2016-10-13:
I would love to see this integrated Chris!

+1

cdjackson, 2016-10-13:
This is in the new code so will be added when it gets merged.

bodiroga, 2016-10-13:
Awesome, I will test it and report back as soon as it is merged, great news! Thanks for your work Chris!

One quick question: how are we going to distinguish between nodes from different controllers in the Things menu of HABmin?

cdjackson, 2016-10-13:
I’m not sure at the moment - it might need a little more thought still...

brononius, 2016-10-16:
For the moment, I'm running 2 USB GEN5 sticks, and no real issues. 
Still some things to solve, but is more related to migration between OH1 and OH2 (I think). 

I'm just using the item and sitemaps files, not really HABmin or so. 
But let me know if I can help/test in anyway!

cdjackson, 2016-10-16:
Thanks for the feedback.

Just be aware that this issue exists. I don't think it will cause any major problems since the binding does some sanity checks on the files when they are read back, so it shouldn't cause incorrect information from one network to be included into a node on another network.

I have implemented this in my refactoring binding, so it's just a matter of time till this is merged.

cdjackson, 2017-02-01:
This is implemented in the development branch so closing.

closed 2017-02-01 by cdjackson