openhab/org.openhab.binding.zwave #142: ZRC-90AU

Gridlok, 2016-09-30:
Hi, 

I spent a good couple of days reading the instructions about the z-wave devices database and am still scratching my head. 

I have a ZRC-90AU which is the Australian version of the ZRC-90. From what I can tell everything is correct in the database with the ZRC-90 except the Type id is 0002, not 0001.

I don't have access to modify the database yet, I have requested it but haven't been approved. But assuming I had write permissions, what would be the correct thing to do here with this? Do I just duplicate the ZRC-90 xml and change it to 0002 and then add it back in as ZRC-90AU? Or is there a way to add it into the ZRC-90 so it is both 0001 and 0002 as they are pretty much the same thing?

And once it is back in the database, do I have to recompile the binding or use a snapshot post its addition to take advantage of it? Or is there a way to test it without doing that?

I've been using z-wave with openhab for years, hard to believe this is the first time this has come up for me.

Cheers,

LJ

mhilbush, 2016-09-30:
In the `References` field, you would change this `0001:8510` to this `0001:8510,0002:8510`.

When Chris approves the change, it will be picked up in his next build.  At that point, you would need to pull the latest snapshot of the zwave binding containing that change.

If you'd like, I can edit the database with this change.

Gridlok, 2016-09-30:
I would very much like that, I don't have access to edit the DB yet.

I thought the change would be something like that, but wasn't quite sure.

Thanks a million, I really appreciate the assistance.

mhilbush, 2016-09-30:
No problem.  I just made the change.

cdjackson, 2016-09-30:
I’m in the USA at the moment and I don’t seem to be able to access my VPN so will update the binding when I get back to the UK (hopefully tomorrow if I can find a flight that doesn’t get cancelled!).

Gridlok, 2016-09-30:
Thanks Chris, I hope the flight goes/went well.

Gridlok, 2016-10-02:
Well, I added in the new binding and I got the same message.

NODE 27: Device discovery could not resolve to a thingType! 5254:0002:8510::1.1

Is that 1.1 at the end important? I didn't think it was as I couldn't see anywhere where it fits in the database.

Thanks for your assistance so far

cdjackson, 2016-10-02:
The 1.1 doesn’t matter in this case - it’s the version number and the database is configured for all versions.

Assuming the versions all updated correctly, then I’m not sure why it’s not worked - everything looks ok. I might add some more debug into the binding today and then get you to send me a debug log.

Gridlok, 2016-10-02:
I just figured it out, there was something up with the aptitude upgrade. The zwave binding wasn't updating.

The following was corrupt /var/lib/apt/lists/partial/openhab.ci.cloudbees.com_job_openHAB-Distribution_ws_distributions_openhab-online_target_apt-repo_Packages.gz so I removed it and re-ran the update and upgrade.

It works after that, but it only supports single press, doesn't do double press and press and hold. Is there a way I can change this?

cdjackson, 2016-10-02:
I don’t know what differentiates this - can you provide a log showing the different presses? Normally I think it’s sent as different scene numbers, but maybe it’s different here?

Gridlok, 2016-10-17:
Sorry for the delay, I had an issue where I couldn't get the log into debug. All sorted now. And for the record, these 8 button remotes accompanied by your fantastic work so far have made my system awesome. I now have them everywhere.

So, here is what I have found. Looking at the logs, I get an entry of Received scene 1 at time 0 when I single press button 1. When I double press it I get Received scene 1 at time 3. When I press and hold it I get a  Received scene 1 at time 2 when I push it and  Received scene 1 at time 1 when I release it.

This is the log.

13:42:00.766 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0B 00 04 00 0C 05 5B 03 60 00 01 C0
13:42:00.767 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_47fddfdb_serial_sof changed from 448 to 449
13:42:00.775 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
13:42:00.778 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0B 00 04 00 0C 05 5B 03 60 00 01 C0
13:42:00.781 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0B 00 04 00 0C 05 5B 03 60 00 01 C0
13:42:00.783 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0C 05 5B 03 60 00 01
13:42:00.785 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 12: Application Command Request (ALIVE:DONE)
13:42:00.786 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 12: Starting initialisation from DONE
13:42:00.788 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@5b83a3 already registered
13:42:00.790 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 12: Incoming command class CENTRAL_SCENE
13:42:00.791 [DEBUG] [dclass.ZWaveCentralSceneCommandClass] - NODE 12: Received CENTRAL_SCENE command V1
13:42:00.793 [DEBUG] [dclass.ZWaveCentralSceneCommandClass] - **NODE 12: Received scene 1 at time 0**
13:42:00.794 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
13:42:00.796 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
13:42:00.797 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got a value event from Z-Wave network, endpoint = 0, command class = CENTRAL_SCENE, value = 1
13:42:00.799 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Updating channel state zwave:device:47fddfdb:node12:scene_number to 1 [DecimalType]
13:42:02.257 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 0C 03 80 03 64 1A
13:42:02.264 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
13:42:02.270 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_47fddfdb_serial_sof changed from 449 to 450
13:42:02.272 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 04 00 0C 03 80 03 64 1A
13:42:02.275 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 04 00 0C 03 80 03 64 1A
13:42:02.278 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0C 03 80 03 64
13:42:02.280 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 12: Application Command Request (ALIVE:DONE)
13:42:02.282 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 12: Starting initialisation from DONE
13:42:02.283 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@5b83a3 already registered
13:42:02.285 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 12: Incoming command class BATTERY
13:42:02.287 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 12: Received Battery Request
13:42:02.289 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 12: Battery report value = 100
13:42:02.290 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
13:42:02.292 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
13:42:02.294 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got a value event from Z-Wave network, endpoint = 0, command class = BATTERY, value = 100
13:42:02.296 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Updating channel state zwave:device:47fddfdb:node12:battery-level to 100 [DecimalType]

When I double press it, I get: 

13:44:40.443 [INFO ] [marthome.event.ItemStateChangedEvent] - zwave_serial_zstick_47fddfdb_serial_sof changed from 452 to 453
13:44:40.444 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0B 00 04 00 0C 05 5B 03 62 03 01 C1
13:44:40.462 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
13:44:40.467 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0B 00 04 00 0C 05 5B 03 62 03 01 C1
13:44:40.471 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0B 00 04 00 0C 05 5B 03 62 03 01 C1
13:44:40.475 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0C 05 5B 03 62 03 01
[... cut ...]
13:46:13.345 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
13:46:13.347 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Got a value event from Z-Wave network, endpoint = 0, command class = BATTERY, value = 100
13:46:13.350 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 12: Updating channel state zwave:device:47fddfdb:node12:battery-level to 100 [DecimalType]

Thanks again, If you are ever in Australia, I would love to buy you a beer... or 10.

Gridlok, 2016-10-17:
One extra thing, when I hold the button down I repeatedly get Received scene 1 at time 2 until I let it go and then I get Received scene 1 at time 1.

cdjackson, 2016-11-01:
The scene reports using Central Scene have now been changed to reflect the number of keypresses. Please see #184.

Closing this - if there's anything further please open a new issue (or reopen this).

closed 2016-11-01 by cdjackson