openhab/org.openhab.binding.zwave #146: Philio PSM02-1 Multi-Sensor not recognised zwave

zissouz, 2016-10-02:
Firstly...thanks for all your hard work on this stuff. 

Sensor details: [http://products.z-wavealliance.org/products/821](url)

I had some joy with this in openhab1 and habmin, but I think I may have ended up hacking a file for another sensor to get it to work. I am trying to move to openhab2 and this is the only sensor I cant get working correctly. I have tried copying the (nodenumber).xml file from my openhab(1) installation but it just shows up as an unknown device.

I also have a Philio PST02A Slim Multi-Sensor (which is very similar but gen5) which is recognised fine. Whats the easiest way to get this PSM02-1 working in openhab2?

cdjackson, 2016-10-02:
The PSM02 should be supported in OH2. You can’t however hack XML files like this - the file is generated by reading information from the device so it should create the file on startup (once the binding has read data from the device).

Can I suggest that we discuss this on the forum instead of the issues list as others will be able to help as well then.

cdjackson, 2016-10-02:
Please open a thread on the forum (community.openhab.org) to discuss further.

closed 2016-10-02 by cdjackson