openhab/org.openhab.binding.zwave #151: ZWaveAlarm CC possible bug

pizerg, 2016-10-03:
I was testing Philio PST02-1X family of sensors (A/B/C variants) which support Notification Report V4 and it got stuck in DYNAMIC_VALUES init stage because NOTIFICATION_REPORT was crashing as it was trying to read some non-existent bytes (ZWaveAlarmCommandClass.java line 164 and the ones following).

I checked with Zniffer and it did not complain about the message being incorrect (see attached captures with NOTIFICATION_SUPPORTED_REPORT and with NOTIFICATION_REPORT messages breakdown). 
I also noticed a difference with Zniffer with the notification supported report, I can see the log:

> NODE: NOTIFICATION_SUPPORTED_REPORT reports V1 ALARM support

But then zniffer shows V1 Alarm field to false.

The other option is that maybe the device is not sending the report properly even if it has the ZWave compliant logo.

I had no time to investigate the issue and run through the official zwave specs so I just coded some protection before reading those values, that way I could continue to work with the sensor, it was just a single line of code before all the failed readings:

`if(serialMessage.getMessagePayload().length > offset + 6){ .... }`

Hope this helps to isolate the issue faster.

PS: All tests were performed with security option disabled (security_inclusion = 2)

[image]
[image]

cdjackson, 2016-10-03:
Hi,
Thanks - the Zniffer out put is interesting - is it also possible to get a log from OH? I’d ideally like to see the serial message received for these two events.

Thanks
Chris

pizerg, 2016-10-03:
Hi Chris,

I can replicate the test and generate the log tomorrow. I'll just need you to provide me if possible the log format that you're using (as I'm using the zwave binding outside OH, but the binding code remains almost the same except for the logging and some other minor changes, but if you can provide me the logger format string I will try to produce one that looks exactly the same so you can use your tool to parse it).

Regards,

Adrián

cdjackson, 2016-10-03:
Thanks Adrian,
If I can get the full serial API packet that would be perfect - any format (hex or decimal) I can work with from there…

Eg (for a multi_instance_association report) -: 01 0B 00 04 00 02 05 8E 03 02 07 00 7F

Thanks.
Chris

pizerg, 2016-10-04:
Hello Chris,

Find attached the logs with the matching zniffer captures for those two messages. 
For the AlarmReportV4.txt the last two lines are mine just to indicate that the payload lenth is shorter than V4 expected and in that case I just set (for now) the alarmType to GENERAL(0).

[AlarmTypeSupportedReport.txt](https://github.com/openhab/org.openhab.binding.zwave/files/508117/AlarmTypeSupportedReport.txt)
[image]
[AlarmReportV4.txt](https://github.com/openhab/org.openhab.binding.zwave/files/508118/AlarmReportV4.txt)
[image]

Regards,

Adrián

cdjackson, 2016-10-08:
For the alarm report, was this an unsolicited report, or was there a request from the binding that resulted in this report (ie a GET request)?

pizerg, 2016-10-10:
FInd attached both cases, unsolicited report and report after Alarm GET

[UnsolicitedReport.txt](https://github.com/openhab/org.openhab.binding.zwave/files/518953/UnsolicitedReport.txt)
[image]

[AlarmAfterRequest.txt](https://github.com/openhab/org.openhab.binding.zwave/files/518954/AlarmAfterRequest.txt)
[image]