openhab/org.openhab.binding.zwave #152: Support new Z-wave device Systemair Remotely Controlled Ventilation

espentj, 2016-10-03:
Is it possible to get the Systemair HVAC Z-wave to Modbus adapter supported?

Type / Id: 0139:0001
Manufacturer: 0276
Device manual: http://products.z-wavealliance.org/ProductManual/File?folder=&filename=Manuals/1768/Z-Wave_to_Systemair_HVAC_Adapter_User_Manual.pdf
Link to device in pepper1 database: http://www.pepper1.net/zwavedb/device/975

NB, the device is NODE 9 in my network. However no XML file is generated in /opt/openhab/userdata/zwave/

Debug Log is attached:
[openhab.log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/506766/openhab.log.zip)

espentj, 2016-10-03:
NB, tested with the latest 2.0.0 SNAPSHOT (build 507)

cdjackson, 2016-10-04:
I've created a new DB entry so feel free to start adding the configuration if you like.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/430

Regarding it not finishing initialisation and creating an XML, I had a look at the log and I don't see reason why it's not completing - the log just seems to stop. Are you sure there is nothing else after this?

espentj, 2016-10-04:
I have fired up the openhab2 project on a Mac instead of on the Pine64. I have also given the system more in debug mode. Here are the resulting logs attached. Still Node 9.
[events.log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/509194/events.log.zip)
[openhab.log 2.zip](https://github.com/openhab/org.openhab.binding.zwave/files/509195/openhab.log.2.zip)

cdjackson, 2016-10-04:
Thanks. I’ve never seen this before, but the device is rejecting the requests. I will have to have a think about how to resolve this.

espentj, 2016-10-04:
I´ll pull the build off the build server and test once it is available.

espentj, 2016-10-05:
Now it looks a lot better (Build 511). Here is the new Thingfile, and I have also included the logs.
[LogsAndThingfile.zip](https://github.com/openhab/org.openhab.binding.zwave/files/510094/LogsAndThingfile.zip)

cdjackson, 2016-10-05:
Great - looks good. It might take another iteration or two, but this will allow me to complete the database definition which I'll try and update tonight.

espentj, 2016-10-06:
I have pulled the latest build down (514) and tested a bit more. Device is correctly identified, and I am unable to add it. However I am still not getting any control of the device. Was there more I should do on my end, or was it supposed to be controllable from the OH2 UI now?

cdjackson, 2016-10-06:
I’m not sure why the device can’t be added - that’s quite strange. What actually happens - it goes into the inbox, but then when you click add, nothing happens?

espentj, 2016-10-06:
Sorry, type, it should be "and I am able to add it".

cdjackson, 2016-10-06:
Ah - ok ;)

What are you trying to control exactly? I’ll likely need a log to see what’s happening - the fan might not work as it might need more configuration, but otherwise I would have expected it to work I think.

espentj, 2016-10-06:
Here are some screen shots, along with the log
[image]
[image]
[EventsAnOH2Log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/514288/EventsAnOH2Log.zip)

cdjackson, 2016-10-06:
In the PaperUI control page is there nothing from this device at all?  What does HABmin show? I’ve not looked at the log, but I guess it won’t show anything if you’ve not even got any control widgets…

Otherwise you might need to configure a sitemap I guess...

espentj, 2016-10-06:
Unfortunately it is not visible in the things for the sitemap config.

[image]

cdjackson, 2016-10-06:
I didn’t mean there. If you look in the thing configuration, what does it show? It should provide at least one switch - not sure what it switches, but the device has this channel and it would be interesting to see if it does anything (like turn the ventilation on/off).

espentj, 2016-10-06:
It shows nothing.
[image]

cdjackson, 2016-10-06:
When you updated the binding, did you delete this thing, and add it back again? If not, you need to do this to get OH to re-read the configuration and add the new channels.

espentj, 2016-10-06:
Ok, I have now removed everything from the controller (removed it on the controller, not only from the UI), and done a hard reset of the controller. I have also done a factory reset of the ventilation z-wave adapter, and then paired them again. Still no luck. See the logs for details.
[openhab.log.zip](https://github.com/openhab/org.openhab.binding.zwave/files/514490/openhab.log.zip)

cdjackson, 2016-10-06:
It looks like there are two files for this device now in the database - probably you’re picking up the wrong one - I’ll need to delete the old one...

cdjackson, 2016-10-06:
I’ve just started a new build - try again with the update in 30 minutes or so…

espentj, 2016-10-06:
That did it!
[image]

Thanks a million :)

cdjackson, 2016-10-06:
Great stuff - thanks. I'll close this now, but any further issues, just yell...

closed 2016-10-06 by cdjackson

berland, 2017-03-23:
I miss the ability to set the "thermostat setpoint" for this device (I got it two days ago and setting it up in OpenHAB2 was really plug and play). 

The THERMOSTAT_SETPOINT seems defined, but there is no channel associated to it. I would like to contribute to getting this correctly configured.