openhab/org.openhab.binding.zwave #153: Textual configuration of zwave-things fails due to missing zwave_nodeid allthough provided.

brstein, 2016-10-04:
My Testsetup consists of a Z-Stick (Gen 5) and a Multisensor (Node 2). I try to define both in the file config\things\default.things as follows:

`Bridge zwave:serial_zstick:controller [ port="/dev/ttyACM0", controller_softreset="false", controller_master="true", heal_enable="true", security_networkkey="01 02 03 04 05 06 07 08 09 10 11 12 13 14 15 16" ]`
`Thing zwave:device:controller:mynode2 (zwave:serial_zstick:controller) [ zwave_nodeid="2" ]`

The bridge comes up properly and autodiscovers the node 2 while the second line fails, because the parameter zwave_nodeid is not seen in ZWaveThingHandler.initialize(), resulting in the following errorlog-message:

`[ding.zwave.handler.ZWaveThingHandler] - NodeID is not set in zwave:device:controller:mynode2`

If I include this device via the inbox everything works fine. Further I tried a more verbose thing definition by providing the properties as displayed if the thing is included via inbox:

`Thing zwave:device:controller:node2 (zwave:serial_zstick:controller) [zwave_nodeid="2",zwave_manufacturer="134",zwave_deviceid="100",zwave_devicetype="2",zwave_version="1.6"]`

The problem remains unaffected. 

Please see also [https://community.openhab.org/t/runtime-problem-zwavethinghandler-does-not-see-zwave-nodeid-2-from-things-config-file/13557/2](url)

cdjackson, 2016-10-04:
This is a bug in ESH https://github.com/eclipse/smarthome/issues/2215

kaikreuzer, 2016-10-06:
I do not agree. If the node id is required for the handler to initialize itself correctly (and find out, how to connect to the right device), this must not be a Thing property, but a configuration parameter. If it were defined as such, it would all work as expected.

giig1967g, 2016-10-06:
I agree with Kai. Infact the zwave_nodeid is missing from the xml configuration files as pointed in https://community.openhab.org/t/zwave-things-file-syntax/14325/36

cdjackson, 2016-10-06:
Sorry - I disagree. In my mind, this is a property. It can not be changed - if the user changes it, the system will stop working. In my understanding, properties are things that come from the device - so the model number, the manufacturer etc - the user can't change these so they are defined as properties. The node id is the same in my mind - it can't be changed - ever. This is why I defined this as a property.

The node id is only one piece of informtion that is required for the handler to initialise. The device model information, manufacturer information etc is all used to derive the device information and decide what thing type to use. Does this mean that all this is also configuration data? I believe elsewhere things like manufacturer and model are defined as properties?

> Infact the zwave_nodeid is missing from the xml configuration files

No - it will not be added to the configuration files in any case, even if we agree that it is to be treated as a configuration parameter.

kaikreuzer, 2016-10-06:
I can only repeat myself: configuration parameters are the configuration of the handler to work and not of the end device. You are aware of this as you stated [here](https://www.eclipse.org/forums/index.php/t/1079343/) yourself: "the current thing configuration is not intended to be used for the configuration of the end device, but only the configuration of the thing handler"

> if the user changes it, the system will stop working

Correct. If the user changes the IP address of a hue bridge, his system will stop working. It is the very same effect, nonetheless, the IP address is a config parameter (which tells the Thing handler, where to connect to). In the Zwave case, it would be the config parameter that tells the handler, which node id to access.

> properties are things that come from the device

No, properties are just a bunch of strings that might be worthwhile for UIs to provide some more information about the device. The values can come from anywhere, be it the thing type, the user or the binding (either adding it statically or dynamically after querying a device).

I think we really need to find a way to separate handler configuration from device configuration soon. In the status quo, the node id is a config parameter required for the handler and thus has to be defined in the XML.

cdjackson, 2016-10-06:
You're right, my "properties come from the device" statement was wrong for sure - but I thought that properties were things that cant change. For example, I think in the Hue binding that the model and device information is defined as a property - if I remember correctly there is a "standard" property for this?

It would be good to get a clear distinction about what should be defined as a property and what should be a configuration. I'm not sure if this is documented anywhere, so as I mentioned, my concept was that properties should be information that the user doesn't change. Now of course in the case where DSL is used, everything is set by the user - configuration and properties. So, my distinction is for the UI where I prefer to make some information configurable, and some fixed. Clearly changing this information is bad ;)

> I think we really need to find a way to separate handler configuration from device configuration soon

Agreed.

> In the status quo, the node id is a config parameter required for the handler 

So I should also move all the other model type information in to configuration as well? What about information about channels which are currently defined as properties as well. I provide a map of command classes that help the channel definition - this is all done as properties. Are you expecting that as this is handler information that it all should be defined as configuration? I really hope this is not the case!

> a config parameter required for the handler and thus has to be defined in the XML.

Really? Surely I can use the configuration handler to define this? I currently don't define all my configuration in the XML so why should this have to go in - is this to make the DSL work?

kaikreuzer, 2016-10-06:
> For example, I think in the Hue binding that the model is defined as a property 

Yes, and the handler is completely ignoring it, it is only the UI that displays it as some additional info to the user. And yes, they are not expected to change (although it is not disallowed), i.e. there are no events sent if they do.

> So, my distinction is for the UI where I prefer to make some information configurable, and some fixed. 

You are losing the nice things about config parameters, like types other like String, etc. (wouldn't the node id be preferably an Integer?).
The correct way to indicate that a user must not change it is to use the [readOnly](https://github.com/eclipse/smarthome/blob/master/bundles/config/org.eclipse.smarthome.config.xml/config-description-1.0.0.xsd#L68) flag on the config parameter. Would this be an acceptable solution for you?

> I provide a map of command classes that help the channel definition - this is all done as properties.

Sounds like a terrible ugly workaround for not defining complete thing types in the first place... If you ask me, it should neither be done as property nor as configuration, but it should be the list of channels referencing the respective channel types.

>  I currently don't define all my configuration in the XML 

Sorry, probably a misunderstanding; I was only referring to adding the node id to the config description XML (i.e. its meta-data).

cdjackson, 2016-10-06:
> You are losing the nice things about config parameters, like types other like String

But I don't need "nice things" ;) I only need to store a few bits of static information. I don't really care if it is an integer - most of the information I need is actually a string and teh complexity of the configuration interface just isn't needed. "Nice things" require more definition, so larger XML files to add another 6 to 10 parameters for every device...

> Sounds like a terrible ugly workaround for not defining complete thing types in the first place... If you ask me, it should neither be done as property nor as configuration, but it should be the list of channels referencing the respective channel types.

In fact this isn't such an issue as it is channel properties, not thing properties where this is done. It's not a hack ;). ZWave has hundreds of devices and we need to be able to (statically) configure how the binding interacts with the device.

However, I do use the properties for storing 'device specific hacks'. This is a list of device specific options to work around bugs in devices which is a static configuration for the binding. I wouldn't want to move all this to configuration options!

Some time back (early 2015) we had a discussion about adding properties to channels to allow customisation of channels and it was my expectation that I could use the thing properties for this sort of configuration in the thing scope. Since it seems this was a wrong assumption, where do you propose this is done?

kaikreuzer, 2016-10-11:
> I don't really care if it is an integer

Well, in the moment that you ask people to specify it in a things-file, you might be happy about being able to do some automatic syntax checks on whether they provided correct information.

> so larger XML files to add another 6 to 10 parameters for every device...

So what you are saying is that you do not want to provide any meta-information, but merely implicitly assume that certain properties are there in a given syntax/format? This brings us back to openHAB 1, where the binding configuration is not machine-readable and the only way to figure out how it needs to be composed is to browse some wiki pages. This is the [main issue that led to the introduction of Things](http://docs.openhab.org/tutorials/migration#understanding-things) in the first place.

> discussion about adding properties to channels to allow customisation of channels

Hm, "customization" is to me a sort of "configuration". Wasn't the discussion rather about adding information for the handler to operate, without having users to know about this info at all?

> Since it seems this was a wrong assumption, where do you propose this is done?

I don't think it was a false assumption, but the place for it is the meta-information, i.e. the thing type, not the thing itself. If you know that for a certain thing type you need to take care of some special hacks in your handler implementation, it is absolutely fine to add such hints as properties, just as you do [here](https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/aeon_dsb45_0_0.xml#L19).

But this issue here is about the nodeId and I think this is something very different, because it describes the Thing instance, not its class (type). And since the nodeId depends on the user's setup, this is a piece of information that defines the (user) configuration.

cdjackson, 2016-12-23:
Implemented.

giig1967g, 2016-12-23:
excellent. Will test tonight.
Any special instructions to test it?
If I remember correctly, you were saying that this feature would need some
adjustments.

thanks

cdjackson, 2016-12-23:
It’s only implemented in my development branch at the moment - so you’d need to use that if you want to test this. It will need a complete redefinition of your system as all thing definitions have changed.

giig1967g, 2016-12-23:
ok. So will wait for instructions.

cdjackson, 2017-02-01:
This should be implemented in the development branch - see below for link to test binary.

https://community.openhab.org/t/z-wave-refactoring-and-testing-and-security/21653/

closed 2017-06-24 by cdjackson