openhab/org.openhab.binding.zwave #160: Add Vision ZS6301 to Database

fleischtorte, 2016-10-07:
Hi Chris,

could you add the Vision ZS6301 CO Detector to the Database? 
On http://www.cd-jackson.com my username is fleischtorte, so please unlock my account that i can add the openhab2 xml.

thanks for your great work!
David

cdjackson, 2016-10-07:
Hi,
I’ve updated your access - any problems, just yell.

Cheers
Chris

fleischtorte, 2016-10-07:
Hi Chris,
i uploaded my openhab2 xml and the device manual, is this all i have to do? Thanks for your fast response :)

cdjackson, 2016-10-07:
Yes - that’s fine unless there are any configuration parameters or association groups which will need to be added manually….  After that I’ll move it into the binding tonight...

Cheers
Chris

cdjackson, 2016-10-08:
Was added with https://github.com/openhab/org.openhab.binding.zwave/pull/162

closed 2016-11-12 by cdjackson