openhab/org.openhab.binding.zwave #184: Central Scene notifications should expose keypress information.

asciiphil, 2016-10-17:
In `ZWaveCentralSceneCommandClass.handleApplicationCommandRequest()`, the third octet of the message payload is treated as a time value.  That doesn't match either the now-public documentation or the one scene controller I have available to me (a HomeSeer HS-WD100+).

The documentation says the lower three bits of that octet indicate how a button was pressed (single through quintuple click or held for a time and released).  My scene controller's behavior matches the documentation.  I'd like the keypress information to be exposed somehow, so I can, e.g. have double clicks trigger different actions than single clicks.

asciiphil, 2016-10-17:
I'm not sure what the best way to accomplish this is, as I'm still coming to grips with the system architecture.  It seems that the most natural approach would be to add the keypress information as an additional channel on the Thing, but that doesn't completely reflect the degree to which the key information and scene ID are coupled; in order for a trigger that cares about keypresses to be useful, it needs to get both pieces of information simultaneously.  An automation rule that only triggers on one of the two pieces risks a race condition that results in it getting old data for the other piece.

cdjackson, 2016-10-17:
I don’t really like to separate out the parts of information into multiple channels since this means the user has to put them back together, and it leaves open the possibility that they get out of sync when they are read (as you already indicate)! My personal preference is to add these as extra attributes to the item state, but this is a significant change to the core and needs more discussion. There’s no simple solution to this at the moment short of some data consolidation which is the approach I used in OH1.

cdjackson, 2016-11-01:
The scene number will be a decimal value in the format "scene.key" where key is the following -:

0 - Single Press
1 - Key Released
2 - Key Held Down
3 - Single Press 2 times
4 - Single Press 3 times
5 - Single Press 4 times
6 - Single Press 5 times

closed 2016-11-01 by cdjackson