openhab/org.openhab.binding.zwave #185: Error getting configuration command class

mhilbush, 2016-10-17:
Not sure why, but I started getting the errors below when updating the config of an FGMS001.  I'm running 
`
185 | Active |  80 | 2.0.0.201610152054 | org.openhab.binding.zwave
`
```
2016-10-17 14:52:55.362 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 36: Configuration update received
2016-10-17 14:52:55.420 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 36: Configuration update config_62_2 to 600
2016-10-17 14:52:55.420 [ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 36: Error getting configurationCommandClass
2016-10-17 14:52:55.421 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 36: Configuration update config_42_2 to 600
2016-10-17 14:52:55.421 [ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 36: Error getting configurationCommandClass

```

cdjackson, 2016-10-17:
Take a look in the XML - it probably doesn’t contain this class (for some reason!). Try deleting the XML and see if it comes back correctly, if not, send the logs ;)

mhilbush, 2016-10-17:
Hmm.  It's not completing initialization.  Getting stuck as ASSOCIATION_GET.

Not sure what to make of this.  In the past, I've added these devices successfully.

Here's the log.
[zwave.txt](https://github.com/openhab/org.openhab.binding.zwave/files/534492/zwave.txt)

cdjackson, 2016-10-17:
I’m not sure what’s up.  There’s not been any change to this command class for 3 weeks. The device is acking the message, but then not responding - I’m slightly concerned that something has caused this, but not sure what it can be - clearly the device is working ok, and responding to lots of other messages, so not sure why it’s bumming out on this one.

mhilbush, 2016-10-17:
Yeah, I thought it looked odd since the device responded to many other commands.  I tried power cycling the device, which didn't help.

mhilbush, 2016-10-18:
I excluded the device from the network, then re-included it.  It initialized successfully this time.  No idea why it was having a problem, but it's working now.

closed 2016-10-18 by mhilbush

cdjackson, 2016-10-18:
Strange!