openhab/org.openhab.binding.zwave #188: Fibaro FGS-223 (Double Switch 2) stuck in STATIC_VALUES init stage

pizerg, 2016-10-19:
Tested with latest version of OH2 (2.0.0.b4). The sensor seems to work properly but it's actually stuck on STATIC_VALUES initialization stage, it seems that the sensor is not responding to some commands required during initialization regarding ZWaveAssociationGroupInfoCommandClass.

Find OH2 logs attached (node 5, the others can be ignored)

[fgs223_stuck.txt](https://github.com/openhab/org.openhab.binding.zwave/files/539172/fgs223_stuck.txt)

cdjackson, 2016-10-19:
It looks like the device isn’t responding to a message - I would try to reset it (exclude/include) to see if it works again.

pizerg, 2016-10-20:
It appears is a sensor issue, some times for no reason it simply stops replying to some messages (also note that parameter 53 is bugged as 0 and 32000 values make the sensor report continuously). 

When the sensor replies as expected is added and initialized without a problem.

closed 2016-10-20 by pizerg