openhab/org.openhab.binding.zwave #191: AeoLabs ZW100 MultiSensor 6 acting strange

thopiekar, 2016-10-25:
As my first sensor was working very well (also had some little problems - will talk about them later), I decided to buy two more.
The problem is now that these are acting strange. The first time I set them up I couldn't manage to set my controller to the first association group. Instead they seemed to be binded to my siren (you can guess how much fun it was).
Then I removed both again and started to set one of them properly up.
By default the controller is associated to the group 1. But this no signal was send by the motion sensor (acting like a binary switch). Now back from one week of vacation the controller is not in the association group anymore (?!) and I tried to readd it but still "in progress" for 2 days.
Additionally approx since that time (made a snapshot update) the thing is still at "Node initialising: PING".

Is this one a known bug?

Thanks!

cdjackson, 2016-11-01:
"acting strange" isn't really a good description. I don't have anything really to go on here so will close this. If you can provide logs showing the device waking up and problems with initialisation etc, then please re-open.

closed 2016-11-01 by cdjackson

thopiekar, 2016-11-01:
I'm sorry about this. It is really hard to describe what is happening here.

Would you like to take a look on that using a TeamViewer session? I would prepare everything so you can take a quick look on that. Don't know how handy my share link was in the past, but I could create a new one so, you can browse the new logs locally.

Regards

thopiekar, 2016-11-01:
Still facing the same problem after readding all things.
As I'm in timezone +1.00 now it should be around 12:43..

HABmin says that node 26 is not communicating, but I'm sure it is. Node 32 is correctly not communicating, as no battery or mains are connected.. Just lying around here..

thopiekar, 2016-11-02:
Made today an update and interestingly also offline things are online now?!
In HABmin there is no possibility to configure any parameters. The section is missing on every thing.
When using PaperUI I still see the old default value of 5121.

cdjackson, 2016-11-02:
I’m afraid I don’t really know what is going on with you system.

I’m also confused that you keep saying some value is 5121 on two different issues - if we’re tracking two issues for the same thing, then we’re going to have confusion.