openhab/org.openhab.binding.zwave #193: ZMNHCD1 Qubino Flush Shutter not recognized, although it is in Database

reyem, 2016-10-26:
Just installed a ZMNHCD1 device. The Qubino Flush Shutter should actually be in the database here
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/213

However, something seems to differ. Not even the manufacturer is recognized, when looking at the node in habmin. attached the node xml.
[node19.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/554492/node19.xml.txt)

I use the beta4 of OH2 with a self compile snapshot zwave binding 2.0.0.201609222252. 

If I can help in updating the database, let me know.

cdjackson, 2016-10-27:
> On 26 Oct 2016, at 22:44, Marco Meyer <notifications@github.com> wrote:
> 
> Not even the manufacturer is recognized,

What do you mean that the manufacturer isn’t recognised?

Why not use a snapshot version rather than compile yourself - then at least I can try to support.

reyem, 2016-10-27:
Hi Chris,

> What do you mean that the manufacturer isn’t recognized?
I mean, habmin is not showing me the manufacturer. But this is probably not relevant. Basically it is listed as "Unknown Device" and I cannot use it. 

> Why not use a snapshot version rather than compile yourself - then at least I can try to support.
Actually it is a snapshot (version 2.0.0.201609222252 installed on a OH2-beta4 installation ). I confused this with home kit add-on, wehere I had to compile it myself to get some additional features. Sorry for this. 
As the corresponding db entry was last updated in August and I have a snapshot from late september, I assume the device should be in the database and recognized. I can also try with the latest snapshot to verify this.

If you could give me write permission on the database at cd-jackson.com, I might be able to review and potentially correct the definition there. My username is "Marco"

Thank you for help. 

Marco

cdjackson, 2016-10-27:
I’ve given you access, but it would be good if you could describe the problem. You can’t change the manufacturer in the database and I’m not sure what you think you can fix?

reyem, 2016-10-27:
> but it would be good if you could describe the problem

The problem is, that the device is not recognized after inclusion. I will try to get the debug log after a restart. What else can I do to describe the problem in more details?

> I’m not sure what you think you can fix?

To be honest, me neither 🙂. However, maybe I can contribute some minor fixes, like the too long descriptions or when I see typos etc. but you are right, it will probably not be able to fix this issue that way...

cdjackson, 2016-10-27:
> 
> but it would be good if you could describe the problem
> 
> The problem is, that the device is not recognized after inclusion. I will try to get the debug log after a restart. What else can I do to describe the problem in more details?
> 
A debug log would be useful as it will show the database search.

> I’m not sure what you think you can fix?
> 
> To be honest, me neither 🙂. However, maybe I can contribute some minor fixes, like the too long descriptions or when I see typos etc. but you are right, it will probably not be able to fix this issue that way…
> 
This is of course always welcome, and it would be great it you could, but I was referring to your point about correcting the database - I thought that you’d found an error.

reyem, 2016-10-27:
Here the debug logs. The zip contains the full openhab.log and a filtered version of it, just containing NODE 19 related logs (openhab.NODE19.log)

[openhab_debug_logs.zip](https://github.com/openhab/org.openhab.binding.zwave/files/556979/openhab_debug_logs.zip)

Too me it looks, like the device is properly identified in the database. But then something must be wrong, as habmin shows me still "Unknown Device".

reyem, 2016-10-27:
Hmmm, after surfing the community forum I found this:
https://community.openhab.org/t/zwave-healing-not-working/15159/5

So I clicked delete in habmin on NODE 19, restarted openhab, the node was again detected as new thing. Then I added it, and now the item is correctly recognized. Although couldn't test it yet, the UI provides me all the parameters and info from the database...

reyem, 2016-10-27:
OK, now the device seem to work at first. However, if I now follow the manual of the device to setup "venetian mode" (set parameter 71 to 1, exclude device, include again ), the device does still not list a channel for controlling the slates. But it should according to the manual. 
Also i see in the manual 3 endpoints ( 2 SWITCH_MULTILEVEL, and 1 SENSOR_MULTILEVEL) documented, whereas the DB entry just lists one endpoint. Could it be, that the DB is not correct in this case?

cdjackson, 2016-10-30:
There should be a channel called blind control for controlling the blind? It’s listed in the database, so I’d be surprised if it’s not shown?

You are right that not all endpoints are in the database and we should fix this. Please supply the XML and I’ll take a look. However, I don’t think this should stop things working completely since endpoint 0 will map to the other endpoints (not the position, but the control) so it should do the same thing I think…  Anyway, let’s update with your XML and we see if that fixes it.

reyem, 2016-10-30:
True, in general the blinds can be controlled. However, this is not really useable in automation, as the lammelas angle depends on the previous state. E.g. If the blind position is on 100% (closed) and i set it to 50%, the lamelas will be in a 90 degree angle (open). But if the position is 0% (open) and I set it to 50%, the lamellas will be in a 0 degree angle (closed). As the blinds can also be controlled from the wall switch, this is pretty complex to manage... I was hoping to be able to manage this with the qubino, as the tilt angle can be managed via standard zwave commands, wheras the fibaro implements this via propritary fibaro specific command class. And this code was never ported to oh2 (which i understand from sn engineering point of view).
The XML was already attached to the first post of this issue. Is this sufficient? Can I help with the db entry update? 
Thanks s lot for your great efforts!

cdjackson, 2016-10-30:
Looking through your XML, your device doesn't seem to have multiple endpoints from the device scan. It would be interesting to get a logfile with full initialisation (ie delete the XML and restart the binding).

I also see from your earlier log that the device isn't responding to some parameters which are in the database, and incorrect for at least the version of the device you have. This tells me the database has some incorrect information in it, although I'd like to see why the device doesn't report multi_channel as this is really the crux of the problem.

reyem, 2016-10-31:
OK, I had to delete the node to let recreate the xml. Just removing the xml and restarting openhab did not work. But now attached the debug log ( Filtered to NODE 19 messages ) of the initialization process. Hopefully I catched all.
[NODE_19.openhab2.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/561279/NODE_19.openhab2.log.txt)

cdjackson, 2016-10-31:
Thanks. I’ll take a look at the log.  Just deleting the XML should be fine - when the binding starts, if it can’t read the XML it performs a full initialisation - this is 100% certain.

cdjackson, 2016-11-01:
Timeout is caused by PARAM 86 - I'll look at this later.

reyem, 2016-11-01:
Ok. shortly checking the manual of the device shows that there is no PARAM 86 to configure for this device. However, could this explain the missing endpoints?

cdjackson, 2016-11-02:
It won’t explain any issues with endpoints since endpoints are interrogated early in the process.

reyem, 2016-11-03:
Just for notice. I also asked Qubino support about the missing endpoints. Their answer was:
"Please wait 30s after exclusion (when you set 71 to 1) before next inclusion."
Although I thought I did this, I did for sure not remove the XML fie after exclusion. So when I have some time I will try
1) set paramter 71 to 1 ( actually this is already the case )
2) exclude the device from the network
3) remove the node xml and wait at least 30 seconds
4) enable debug log
5) include again.

Maybe then the endpoint appears in the scan.

Question: What would happen if I manually add the endpoint to the XML?

cdjackson, 2016-11-03:
Strange - not heard of this before!

If you manually add the endpoint, it almost certainly won’t work. The XML reflects (or should reflect!) the configuration of the device, modifying the XML in most cases will just screw things up, not make the device support endpoints that don’t exist (there are exceptions to this, but not many).

reyem, 2016-11-10:
Tried to exclude the device, wait 30 seconds and include it again. What is strange for me, is that after exclusion, I neither put the device nor the controller explicitly in inclusion mode, but when I restart openhab the device shows up as a new things the logs show that the device is scanned and the node xml is created ( still with just endpoint 0 ). Why is the device even scanned after I excluded it? I also tried to exclude it using openhab1 with habmin and after that wait 30 seconds. Then start openhab2.. The Node is automatically included into the zwave network... But Why?
Attached the log files ( full log and filtered for NODE 19 messages)
[logs.zip](https://github.com/openhab/org.openhab.binding.zwave/files/584497/logs.zip)

reyem, 2016-11-10:
OK, after I successfully excluded the device. waited 30 seconds and included again, the node was new initialized and scanned. However. Still just endpoint 0 listed in the Node19.xml. Attached the node xml and the log file...
[node19.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/584538/node19.xml.txt)
[openhab2.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/584539/openhab2.log.txt)

Do you have any suggestions on how to proceed?

cdjackson, 2016-11-10:
Hi Marco,
I’m afraid I’m not sure what’s up. The device isn’t reporting that it supports multi-channel even, so at the moment I don’t know… The only thing I can think of is your device is an old version - I can’t find a manual for your version (version 1.1) - maybe it doesn’t support multiple endpoints?

Maybe you can see if you find get a manual for this software version?

Chris

reyem, 2016-11-11:
Hi Chris,
The printed manual which came with the device is almost similar to the one attached in your database and also documents the additional endpoints... I sent another question to the support of the manufacturer, but did not get any reply yet. In the mean time I will try to see if another controller software shows me the same result. I plan to buy a Z-Way license, as my conroller is a uzb from zwave.me... or do you know a good alternative software for debugging the controller/network?
Thx anyway 
Marco

cdjackson, 2016-11-11:
Trying ZWay is fine - you could try the Zensys tools although I’ve never tried them as they are windows ;)

We could try and force add the multi-channel class, but I’m not convinced it will work given the device isn’t reporting it in the NIF and this is mandatory as defined in the ZWave protocol (and I’ve confirmed this through my Sigma contact).

reyem, 2016-11-14:
Hi Chris,

I failed with Z-Way, and now I will try Zensys ( my mac can run VMs ;-) )...
However, I got an interesting reply from Qubino support which took also a look at this thread. The relevant part of their answer was:

> You are experiencing this problem, because the OpenHAB platform, apparently, doesn't take into account the use of dynamic endpoints (endpoints, which are enabled manually). Dynamic endpoints are described in the official Z-wave specifications in the following document (section 4.84.1): http://z-wave.sigmadesigns.com/wp-content/uploads/2016/08/SDS12657-12-Z-Wave-Command-Class-Specification-A-M.pdf 

What do you think? Is this assumption right that the zwave binding does not support dynamic endpoints yet? If so, I guess we should make a feature request out of this issue!?

I will give you some update, once I have tested the device using Zensys.

Marco

cdjackson, 2016-11-14:
Interesting. I've not looked at dynamic end points but it is true that it is not implemented (mainly as I've not seen any device that implements them). 

I will take another look at the log to see what it shows about dynamic end points in the endpoint report.

cdjackson, 2016-11-15:
The problem I have with this is that the device does not indicate that it supports multi-channel at all - ie it is not indicated in the NIF. Therefore, OH does not interrogate the multi-channel command class, and therefore doesn't find any endpoints. This is not related to dynamic endpoints as far as I can see since the documentation indicates that a device that is controllable should directly - from my reading of the doc, the device should say that it supports MULTI_CHANNEL in the NIF, and if an endpoint is created/changed, it should send the CAPABILITY_REPORT - and I see neither.

I therefore don't see how I'm meant to know that the device is supporting MULTI_CHANNEL if it doesn't indicate this somehow, but I might be missing something in the spec.

The following shows the NIF from one of your logs...

[image]

The binding only looks for multi channel encapsulation if this is reported in the NIF.

cdjackson, 2016-11-15:
Closing the loop...

https://community.openhab.org/t/unable-to-rotate-venetian-blinds-via-rules-z-wave-qubino-flush-shutter/16690/19

reyem, 2016-11-15:
As soon as I find time I will play around with the device using Zensys. I will update you if it finds the endpoints there and allows me to control these...

reyem, 2016-11-16:
OK, some findings.
1) I tried with Zensys and the MULTI_CHANNEL is also missing after I followed the exact description of the support team.
2) The support team confirms that the MULTI_CHANNEL should be in the NIF when the node is included again. So I guess you interpret the spec properly
3) I found a node xml from the point in time when I started trying to get this running. The first node xml I have, actually has following in, which does not appear in later versions of node xml anymore:
`<commandClass>MULTI_INSTANCE</commandClass>`
I assume this is what you are looking for!? Or is MULTI_INSTANCE, MULTI_CHANNEL not the same?  

Attached the version of the node xml which I assume would be the correct one.
[node19.xml.possible_correct.txt](https://github.com/openhab/org.openhab.binding.zwave/files/594061/node19.xml.possible_correct.txt)

4) However, there is definitely something wrong with the device, as I cannot get this MULTI_INSTANCE report anymore in the NIF... Also not with Zensys!? I'm on this with the Qubino support team now.

cdjackson, 2016-11-16:
> So I guess you interpret the spec properly

Good to know :)

> I assume this is what you are looking for!? Or is MULTI_INSTANCE, MULTI_CHANNEL not the same?

Yes this is fine. While MULTI_INSTANCE and MULTI_CHANNEL are in one respect different, they are still the same class - MULTI_INSTANCE is V1, and at V2 Sigma changed the name to MULTI_CHANNEL. openHAB currently still calls this MULTI_INSTANCE, but don't confuse this with the fact that MULTI_CHANNEL isn't supported - it is - it's just that the name in the XML still uses the old name. This is also true of the ALARM class which was renamed (at V3 I think) from ALARM to NOTIFICATION... It's just a name, and doesn't affect the implementation...

The XML looks fine, but doesn't help if the NIF doesn't reflect this when you run it. In the forum link I posted above there are some people saying it works ok, and some not. I think that there are problems with the switching of this with the "switch parameter 71 (?) and exclude and re-include" process. It seems overly complex and unreliable.

> I'm on this with the Qubino support team now.

Thanks. I think they are keen to say that the problem is OH doesn't support MULTI_CHANNEL so you might want to point them at the above explanation. Also the forum post referenced above that shows that this device does work on OH - if MULTI_CHANNEL is enabled in the device which seems to be the issue (in my mind at least ;) ).

reyem, 2016-11-17:
Bingo! After the last try I managed to get the MULTI_INSTANCE/CHANNEL in the NIF ( I'm guessing ). Attached the Log File:
[NODE_24.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/598632/NODE_24.log.txt)
and the node xml file
[node24.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/598633/node24.xml.txt)

From my layman eyes everything looks as expected now. Also the XML lists now 2 endpoint. But openhab2 does still not show me channels for these endpoints, respectively just one channel for a multi level switch, but I should get two. Can this now be fixed through the database?

P.S. For anybody reading this thread and having troubles with Parameter 71 and this device... I feel with you ;-). Here how I managed to get it working.
PreCondition: Param 71 set to 1, but NIF does not report MULTI_INSTANCE
1) set Param 71 to 0 -> save ( make sure it has been set really )
2) set Param 71 to 1 -> save ( make sure it has been set really )
3) exclude device from network and delete Node xml, stop open hab
4) switch off power supply of the device ( according to support this is an alternative to wait 30s )
5) switch on power supply again, start openhab
6) Include device to the network
7) Add Thing to openhab via habmin
8) Be patient until the binding until processed all your nodes again

cdjackson, 2016-11-19:
I have updated the database with the above process...
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/213#overview

cdjackson, 2016-11-19:
Database is updated - I'll export later today. It will contain duplicate channels which we should eliminate once we confirm this is working. (ie there will be a switch and switch 1 which I think will be the same since the device duplicated endpoint 1 as the root device)

reyem, 2016-11-19:
Thank you a lot! I wanted to try it today. Unfortunately the newest snapshot of the binding seems not to be compatible with the beta4 release anymore. There were plenty of errors in the log and no node came online anymore. So I will have to do it, when I have time to switch to a snapshot release of OH2. Just switching the sources.list for apt-get seems not to work. So this will take longer :-(

What I recognized in the database is, the endpoint 3 would still be missing. endpoint 3 is just available if a temparture sensor is connected to the module. The manual documents all command classes for the endpoint etc. This just as a hint. Maybe it would be worth to alrady add this endpoint as well.

cdjackson, 2016-11-20:
Hi,
Sorry about the change - this was something that changed in ESH and nothing to do with me ;)

This should be compiled into the binding this morning (EU time)…

Chris

reyem, 2016-11-23:
Hi Chris,
Finally I updated to the latest snapshot. I can confirm, the module works now. I get the endpoints and can control shutter position and lamella tilt angle separately via dimmers. Endpoint1 is as you expected the same as the main control channels. However the main channel seems to be exposed as "blind control", whereas endpoint 1 and 2 are exposed as "dimmer". So by removing endpoint1, you would remove a dimmer channel. Not sure if this makes any difference though. Here the screenshot of the channels in paper ui:
[image]

Thanks a lot for your support!
Marco

closed 2016-11-23 by reyem