openhab/org.openhab.binding.zwave #197: Possible database update for Horstmann Controls SIR-321

tristann9, 2016-10-30:
Hiya

I have a slightly updated SIR-321 and OpenHab isn't able to recognise it.  The details on my controller are showing up as (0059:0010:0002:1.0) and the file horstmann_sir321_0_0 specifies 0001 and not 0002.  I've made the following change on horstmann_sir321_0_0.xml:

      <property name="manufacturerRef">0010:0001,0010:0002</property>

I then built the jar file and deployed it, but no luck.  This is my first time updating the device details so not sure if I'm doing it all wrong.  Is there anything else I need to do to reflect the update?

Cheers.

cdjackson, 2016-10-30:
Please can you confirm that your version has the same parameters as in the database -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/313 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/313>

Also please provide the XML openhab has created for your device.

tristann9, 2016-10-30:
Thanks for your quick reply.

From quickly glancing, the features look identical.  However, I'll give it a much closer look tomorrow to see if I find any discrepancies between both versions.

tristann9, 2016-10-30:
Just an update, reading through the manual it states "All command classes are version 1 unless
otherwise stated.".  The only command showing as v2 is labeled as "Manufacturer Specific" with the description "Secure Controls (UK) Manufacturer ID, RFModule Serial Number".

Apart from that, I'm able to see the device on openhab with my manufacturerRef changes I mentioned aslier.  Not sure why it took a while to pick up my changes even after restarting openhab, but it's picking them up now so to answer your question I think it should be safe to say that version two has the same parametres as version 1 (shown in the database).  The only exception in regards to the database is that it does not include the Command Class "Schedule", which permits the countdown timer to countdown the time the device turned on.  More details regarding the schedule command class are seen on page 17 from the manual http://www.securemeters.com/files/4014/1317/3487/SIR_321_manuals.pdf.

cdjackson, 2016-10-31:
Ok - thanks. I think for now at least it seems ok to make them the same. I’ll update the database to add your device IDs.

The schedule command class isn’t supported at the moment anyway…

cdjackson, 2016-11-01:
I think this should now be fixed in todays binding.

tristann9, 2016-11-02:
Cheers.  I've also submitted a new merge request to support the SES external probe.

closed 2016-11-02 by tristann9