openhab/org.openhab.binding.zwave #198: AeoLabs ZW100: Incorrect range for temperature parameter

thopiekar, 2016-10-31:
In our database we've got a range set up from 0...100 and a default of 20.
That correct so far, BUT: The unit of the value is encoded into the parameter, too.

For example:
Here my value is "5121", which is in hex "0x1401". So in my case I have a hysteresis of 1.4 in the unit "°C" like given by the last two numbers "01". In Fahrenheit you will need to use "02" instead.

So assuming that 100 is the maximum here (have not checked the docs) and we use Fahrenheit the highest value we can set here is truely: "0X10002" -> "65538" and the lowest "0x001" -> "1".

At the moment I would correct it like that, but would like to keep this issue report as a reminder that we probably need something like formulas and special widgets to decode/encode values here..

* http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/73

cdjackson, 2016-10-31:
What parameter are you talking of?

thopiekar, 2016-10-31:
Sorry, I'm talking about parameter 41: "Temperature Threshold"

cdjackson, 2016-10-31:
I’ll need to look at this more - there are multiple versions of this device and I think you are talking only about a recent version. The database can probably be designed to split out the unit and value into separate parameters using the bitfield.

I think you really mean this device in the database -:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/355 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/355>

thopiekar, 2016-10-31:
See no difference in both entries at Type/ID... However my device here is: "Type / ID 0002:0064".

From the summary of 73: `0002:0064,0102:0064,0202:0064`
From the summary of your 355: `0002:0064,0102:0064,0202:0064`

Probably the reason why I experience #191? Or is there another way how we identify the newer one from the older one? (Just checking here..) :thinking:

thopiekar, 2016-10-31:
Oh, just added my third ZW100 10minutes before and my first ZW100 just dropped into: `Node is not communicating with controller`.
Wired things are happening here...

cdjackson, 2016-10-31:
The difference is the version number. The older version doesn’t encode parameter 41 like you said - it’s 2 bytes and doesn’t include the unit. The newer version (1.7) adds the unit and makes the length 3 bytes.

I’m not really sure why you link this issue to the other one - they look totally different???

thopiekar, 2016-10-31:
Just had the idea that the ZWave binding is identifying my ZW100 as the old one, because both have the same ID and the effects of the other issue I reported is also affected by this behaviour. But that's obviously wrong, sorry.

cdjackson, 2016-10-31:
What version do you have then? The binding doesn’t just use the ID to decide what device you have - it uses version numbers too.

thopiekar, 2016-10-31:
[image]

thopiekar, 2016-10-31:
When talking about the "version" you mean "firmware version"?

cdjackson, 2016-10-31:
Yes.

cdjackson, 2016-11-01:
I've updated http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/355 to add two new parameters to separate out the threshold and the unit. This will be in the next version (hopefully tomorrow) - feedback welcome.

thopiekar, 2016-11-01:
Yes, of course!

mhilbush, 2016-11-01:
@cdjackson  Can you clarify what is needed for existing ZW100 things?  Do they need to be deleted and rediscovered?

cdjackson, 2016-11-01:
No - for parameter description updates there's nothing to do - other than to update the binding of course.

Note that this is only for V1.7 at the moment - I've not found a full description of 1.8 - this can probably be added there at some point (assuming it works ok ;) ).

thopiekar, 2016-11-02:
The value is still "5121"

cdjackson, 2016-11-02:
> On 2 Nov 2016, at 18:04, Thomas Karl Pietrowski <notifications@github.com> wrote:
> 
> The value is still "5121"

What value is 5121?

thopiekar, 2016-11-02:
The old value, which is here not decoded.. Like described in the first post: "5121" -- in hex  --> "0x1401" --> 1.4°C for parameter 41: "Temperature Threshold"
I expected to see 1.4°C here after the update.

cdjackson, 2016-11-02:
You won’t see 1.4degC for sure - never.

The original parameter should be hidden, but I think I forgot to do this. You should however have two separate parameters - one that shows degC and one that shows 14.

cdjackson, 2016-11-02:
Actually this might not be merged yet - it doesn’t seem to be in the GH version.

thopiekar, 2016-11-13:
I've reinstalled a new OH2 snapshot, set up everything again and the problem does not exist anymore.

closed 2016-11-13 by thopiekar