openhab/org.openhab.binding.zwave #205: Please add device Popp outdoor switch device

mkotsbak, 2016-11-04:
Please add this Popp device: http://www.popp.eu/products/actuators/wall-plug-switch-outdoor/

See xml here: https://github.com/openhab/openhab2-addons/files/570185/node11.xml.txt

Manual: http://www.popp.eu/wp-content/uploads/2016/01/Manual_IP44_POPP_En.pdf

cdjackson, 2016-11-04:
This seems to be the ZWave.me device http://www.cd-jackson.com/index.php/zwave/zwave-device-database/database-history/devicesummary/142 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/database-history/devicesummary/142>

I’ve added your XML and will update the OH2 binding with this device tonight.

mkotsbak, 2016-11-04:
It seems mostly correct, but parameter 1: "Set LED indication mode" is wrong.

Manual says:

```LED mode (Parameter number 1, Size 1)
Set LED indication mode
Value Description
0 Disable
1 Show switch (default)
2 Operated by Indicator Command Class```

mkotsbak, 2016-11-04:
Btw, it appears correctly in open zwave control panel, so it must be correctly registered somewhere...

cdjackson, 2016-11-04:
Ok - so I think we should probably update the database since the device seems to be the correct one. If there are multiple versions then we’ll have to manage this later although it looks like the only difference is the option 2 wording...

cdjackson, 2016-11-04:
What do you mean by "it appears correctly in open zwave control panel, so it must be correctly registered somewhere…”

mkotsbak, 2016-11-04:
I mean in some database it is correctly registered, or does it come from the device itself?

Found this in some xml file inside open-zwave-control-panel:

```
                                <Value type="list" genre="config" instance="1" index="1" label="Led mode" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min=
"0" max="0" vindex="1" size="1">
                                        <Help>Set LED indication mode</Help>
                                        <Item label="Disable" value="0" />
                                        <Item label="Show switch" value="1" />
                                        <Item label="Indicator CC" value="2" />
                                </Value>
```

cdjackson, 2016-11-04:
Do you mean that this option is correct in the OZW database? If so, then thats normal as it’s a totally different database - if someone types wrong data into a database, then it will be wrong. The OZW database is not linked to the OH database in any way, so 100% for sure, they will be different.

mkotsbak, 2016-11-04:
Ah ok. I thought they were linked/imported in some way.

cdjackson, 2016-11-04:
No - OH has no link in any way to OZW - code or database, or....

mkotsbak, 2016-11-07:
Works fine with the latest OH2 snapshot release :)

closed 2016-11-07 by mkotsbak