openhab/org.openhab.binding.zwave #210: Monoprice Garage Door Report Alarm vs Sensor

ptmuldoon, 2016-11-06:
As mentioned on the OpenHab Forums:

https://community.openhab.org/t/zwave-garage-door-tilt-sensor-status/16226/4  and here
https://community.openhab.org/t/monoprice-garage-door-sensor-alarm-sensor-switched/16254/3

The Tilt Sensor appears to be reporting the ON|OFF value under the Alarm setting vs the Sensor setting,  It also appears the Battery Level is not showing, but have seen anything in the logs to indicate why that would.

The MonoPrice Garage Door Tilt Sensor is being reported as the ZGB101 Sensor by Vision Security here in the database:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/103

And the the log show the device initialing ok but debug Zwave info on the device shows as follows the alarm as the On|Off

[code]
2016-11-06 07:40:14.449 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising Thing Node...
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising channel zwave:device:158135ef039:node17:sensor_binary
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising cmd channel zwave:device:158135ef039:node17:sensor_binary
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising poll channel zwave:device:158135ef039:node17:sensor_binary
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising state channel zwave:device:158135ef039:node17:sensor_binary
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising channel zwave:device:158135ef039:node17:alarm_burglar
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising cmd channel zwave:device:158135ef039:node17:alarm_burglar
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising poll channel zwave:device:158135ef039:node17:alarm_burglar
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising state channel zwave:device:158135ef039:node17:alarm_burglar
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising channel zwave:device:158135ef039:node17:battery-level
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising cmd channel zwave:device:158135ef039:node17:battery-level
2016-11-06 07:40:14.450 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising poll channel zwave:device:158135ef039:node17:battery-level
2016-11-06 07:40:14.451 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Initialising state channel zwave:device:158135ef039:node17:battery-level
[/code]

[code]
[DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Polling...
2016-11-06 08:10:14.464 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Polling zwave:device:158135ef039:node17:sensor_binary
2016-11-06 08:10:14.464 [DEBUG] [converter.ZWaveBinarySensorConverter] - NODE 17: Generating poll message for SENSOR_BINARY, endpoint 0
2016-11-06 08:10:14.465 [DEBUG] [dclass.ZWaveBinarySensorCommandClass] - NODE 17: Creating new message for application command SENSOR_BINARY_GET
2016-11-06 08:10:14.465 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Polling zwave:device:158135ef039:node17:alarm_burglar
2016-11-06 08:10:14.465 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 17: Generating poll message for ALARM, endpoint 0, alarm BURGLAR
2016-11-06 08:10:14.465 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 17: Creating new message for application command NOTIFICATION_GET V2
2016-11-06 08:10:14.465 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Polling zwave:device:158135ef039:node17:battery-level
2016-11-06 08:10:14.465 [DEBUG] [rnal.converter.ZWaveBatteryConverter] - NODE 17: Generating poll message for BATTERY endpoint 0
2016-11-06 08:10:14.465 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 17: Creating new message for application command BATTERY_GET
[/code]

[code]
[DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 17: Application Command Request (ALIVE:DONE)
2016-11-06 08:40:31.622 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 17: Starting initialisation from DONE
2016-11-06 08:40:31.622 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@3f7d2729 already registered
2016-11-06 08:40:31.622 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 17: Incoming command class ALARM
2016-11-06 08:40:31.622 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 17: Received ALARM command V2
2016-11-06 08:40:31.622 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 17: Process NOTIFICATION_REPORT V2
2016-11-06 08:40:31.622 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 17: NOTIFICATION report - 0 = 0, event=2, status=255
2016-11-06 08:40:31.622 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 17: Alarm Type = BURGLAR (0)
2016-11-06 08:40:31.622 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
2016-11-06 08:40:31.623 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2016-11-06 08:40:31.623 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
2016-11-06 08:40:31.623 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2016-11-06 08:40:31.623 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 2, type OnOffType
2016-11-06 08:40:31.623 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Updating channel state zwave:device:158135ef039:node17:alarm_burglar to ON [OnOffType]
2016-11-06 08:40:31.623 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Immediate, dest=17, callback=230, payload=11 04 71 04 00 07 
2016-11-06 08:40:31.623 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 11 09 71 05 00 00 00 FF 07 02 00 
2016-11-06 08:40:31.623 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=ApplicationCommandHandler, callback id=230, expected=ApplicationCommandHandler, cancelled=false        transaction complete!
2016-11-06 08:40:31.623 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
2016-11-06 08:40:31.623 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2016-11-06 08:40:31.623 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 1. Queue={}
2016-11-06 08:40:31.624 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 17: Response processed after 104ms/1434ms.
2016-11-06 08:40:31.624 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
2016-11-06 08:40:31.624 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 11 02 80 02 25 E7 B6 
2016-11-06 08:40:31.624 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 17: Sending REQUEST Message = 01 09 00 13 11 02 80 02 25 E7 B6 
2016-11-06 08:40:31.648 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8 
2016-11-06 08:40:31.652 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-11-06 08:40:31.652 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8 
2016-11-06 08:40:31.652 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8 
[... 4 lines cut ...]
2016-11-06 08:40:31.686 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 E7 00 00 03 00 00 01 
2016-11-06 08:40:31.686 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 13 E7 00 00 03 00 00 01 
2016-11-06 08:40:31.686 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=E7 00 00 03 
[... cut ...]
2016-11-06 08:40:31.710 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
2016-11-06 08:40:31.710 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
2016-11-06 08:40:31.710 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Got a value event from Z-Wave network, endpoint = 0, command class = BATTERY, value = 100
2016-11-06 08:40:31.710 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 17: Updating channel state zwave:device:158135ef039:node17:battery-level to 100 [DecimalType]
[/code]

SuperOok, 2017-09-03:
I encounter the same issue after an Upgrade from OH 2.0 to 2.1: I only see the Burglar-Alarm channel triggered, but not the sensor channel:

```23:50:44.694 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 06 0A 71 05 07 FF 00 FF 07 03 00 00 90
23:50:44.700 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
23:50:44.703 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 10 00 04 00 06 0A 71 05 07 FF 00 FF 07 03 00 00 90
23:50:44.706 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 10 00 04 00 06 0A 71 05 07 FF 00 FF 07 03 00 00 90
23:50:44.709 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 06 0A 71 05 07 FF 00 FF 07 03 00 00
23:50:44.711 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 6: Application Command Request (ALIVE:DONE)
23:50:44.713 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 6: Starting initialisation from DONE
23:50:44.715 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@8dc2e3 already registered
23:50:44.716 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 6: Incoming command class ALARM
23:50:44.718 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 6: Received ALARM command V2
23:50:44.720 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 6: Process NOTIFICATION_REPORT V2
23:50:44.722 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 6: NOTIFICATION report - 7 = 255, event=3, status=255
23:50:44.724 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 6: Alarm Type = BURGLAR (7)
23:50:44.725 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
23:50:44.727 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Got an event from Z-Wave network: ZWaveAlarmValueEvent
23:50:44.729 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
23:50:44.731 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
23:50:44.733 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
23:50:44.735 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating channel state zwave:device:96c48a58:node6:alarm_burglar to ON [OnOffType]
23:50:44.746 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 8: Transaction not completed: node address inconsistent.  lastSent=8, incoming=255
```

* Node 6: ZG8101 Garage Door Tilt Sensor
* Node 8: PAN04 In Wall Dual Relay(1 Way) Switch Module 2x 1.5kW with power meter

Every time I move the sensor, the sequence above is shown, including the last line about NODE 8. Might there be some conflict or something alike?

Any further information I could send?

SuperOok, 2017-09-03:
After wild googling and trial-and-error I found this thread:
https://community.openhab.org/t/zwave-vision-garage-door-detector-zg8101-doent-work-with-oh2/24934

It states that the type of the binary sensor was changed to door-sensor which seems to imply that a contact Item is used.

Hence, I changed my Item type from Switch to Contact and correspondingly changed all rules from ON/OFF-Type to OPEN/CLOSED. And now... it just works.

It just surprises me that I only changed stuff in the Item-configuration and nothing in the zwave configuration. So my problem was that no Zwave telegram was reported at all when using a Switch item in the debug log file. So my wrong conclusion was that there must be smth wrong in the Zwave config/database and not in the downstream Items config. 
So might that be a bug that there is no debug info about the sensor_door channel when linked to a Switch item?

cdjackson, 2017-12-30:
> So might that be a bug that there is no debug info about the sensor_door channel when linked to a Switch item?

The binding doesn't know about items - just channels. If you link an invalid item to a channel, then the binding can't resolve it.

It looks like this is working, so closing.

closed 2017-12-30 by cdjackson