openhab/org.openhab.binding.zwave #213: Fortrezz SSA1/2 EU version

Rallymen007, 2016-11-09:
Hello,

Current device is:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/220

My device has a slightly different ID:
0084:0311:010B::1.11

I guess it's because it's the EU version. The associated XML is the same, as I've been able to test this by adding my ID to the detection list for the current SSA1/2.

I don't have access to the DB so I can't do the edit myself. My account on the DB is Rallymen007, like on GH, so you could give me edit access so I can do it myself, or just add the line and we'll be done.

Thanks

cdjackson, 2016-11-09:
I've added it to the database - will export in the next day.

Rallymen007, 2016-11-18:
Got a nightly build, tested my device and it works as expected. Thanks!

closed 2016-11-18 by Rallymen007