openhab/org.openhab.binding.zwave #216: Wrong Wakeup Configuration Value reported in HABmin

Alessandro65, 2016-11-13:
In openHAB 2.0 Build #574, changing the Wakeup Configuration Value in HABmin for a Danfoss LC-13 thermostat the value is updated in HABmin interface and the device behaviour is modified accordingly. BUT after a openHAB restart the Wakeup Configuration Value displayed in HABmin is the default value, even if the device behaviour shows that the correct (previous) value has been retained.

Changing the Wakeup Configuration Value in HABmin, the interface showed PENDING until the value was transferred to the device, after this the HABmin interface value matched with the device behaviour (as can be verified in the debug log). The problem is that after a openHAB restart the value was (correctly) retained by the device but not transfered back in openHAB/HABmin 

The XML "node" stored in "/var/lib/openhab2/zwave" shows:
```
 <entry>
  <commandClass>WAKE_UP</commandClass>
  <WakeUpCommandClass>
    <version>2</version>
    <instances>1</instances>
    <versionSupported>2</versionSupported>
    <targetNodeId>1</targetNodeId>
    <interval>300</interval>
    <minInterval>60</minInterval>
    <maxInterval>1800</maxInterval>
    <defaultInterval>300</defaultInterval>
    <intervalStep>60</intervalStep>
    <lastWakeup>2016-11-12 11:36:50.174 UTC</lastWakeup>
    <isGetSupported>true</isGetSupported>
  </WakeUpCommandClass>
</entry>
```
I.E. the interval is 300 even if the effective (device) value is 900

For further detail refer to this thread:
https://community.openhab.org/t/wakeup-configuration-value-not-updated-in-habmin/16626/8

cdjackson, 2016-11-15:
Hopefully this should be fixed as the XML is now saved on closing of the binding so should be ok unless there is a crash.

cdjackson, 2016-11-22:
Closing this as it should be fixed - if further issues are experienced then please reopen.

closed 2016-11-22 by cdjackson