openhab/org.openhab.binding.zwave #217: XML Document "cannot be parsed" warnings

BClark09, 2016-11-13:
* OH2 Version: \#586 Online Snapshot
* Binding Version: 2.0.0.201611122320

For most popp devices and one kwikset device in the database, a warning in the OH console comes up for each device, regardless of them being apart of the system:

```
08:56:27.670 [WARN ] [ig.xml.osgi.XmlDocumentBundleTracker] - The XML document '/ESH-INF/thing/popp_004407_1_0.xml' in module 'org.openhab.binding.zwave' could not be parsed:  : ParseError at [row,col]:[61,37]
Message: The entity name must immediately follow the '&' in the entity reference. :  : ParseError at [row,col]:[61,37]
Message: The entity name must immediately follow the '&' in the entity reference.
---- Debugging information ----
message             :  : ParseError at [row,col]:[61,37]
Message: The entity name must immediately follow the '&' in the entity reference.
cause-exception     : com.thoughtworks.xstream.io.StreamException
cause-message       :  : ParseError at [row,col]:[61,37]
Message: The entity name must immediately follow the '&' in the entity reference.
class               : org.eclipse.smarthome.config.xml.util.NodeValue
required-type       : org.eclipse.smarthome.config.xml.util.NodeValue
converter-type      : org.eclipse.smarthome.config.xml.util.NodeValueConverter
path                : /thing-descriptions/thing-type/properties/property
line number         : 61
class[1]            : java.util.ArrayList
converter-type[1]   : com.thoughtworks.xstream.converters.collections.CollectionConverter
class[2]            : org.eclipse.smarthome.config.xml.util.NodeList
converter-type[2]   : org.eclipse.smarthome.config.xml.util.NodeListConverter
class[3]            : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult
converter-type[3]   : org.eclipse.smarthome.core.thing.xml.internal.ThingTypeConverter
class[4]            : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionList
converter-type[4]   : org.eclipse.smarthome.core.thing.xml.internal.ThingDescriptionConverter
version             : 1.4.7
-------------------------------
```

It has a problem with the "&" symbol used in each of these files:

```
/ESH-INF/thing/popp_05438_0_0.xml
/ESH-INF/thing/popp_005107_0_0.xml
/ESH-INF/thing/popp_012501_0_0.xml
/ESH-INF/thing/popp_004001_0_0.xml
/ESH-INF/thing/popp_123610_0_0.xml
/ESH-INF/thing/popp_009303_0_0.xml
/ESH-INF/thing/popp_123580_0_0.xml
/ESH-INF/thing/popp_009402_0_0.xml
/ESH-INF/thing/popp_123665_0_0.xml
/ESH-INF/thing/popp_009204_0_0.xml
/ESH-INF/thing/kwikset_914trl_0_0.xml
/ESH-INF/thing/popp_004407_1_0.xml
```

cdjackson, 2016-11-13:
Thanks. I’ve fixed this in the export function in the database and will do a complete export today/tonight which should resolve all these errors...

BClark09, 2016-11-13:
Thanks @cdjackson!

closed 2016-11-13 by cdjackson