openhab/org.openhab.binding.zwave #221: Add Merten 5051xx, Merten 5052xx, Merten 507001

Arsur, 2016-11-14:
Hi
I need some help to get these devices into the database so openhab can find them. 

Merten 5051xx
https://www.merten.de/uploads/tx_seqdownload/V5051_581_01_web.pdf
http://www.pepper1.net/zwavedb/device/68
http://www.pepper1.net/zwavedb/device/68/68-007a-0001-0002-02-02-1b-01-08.xml
Openhab Manu: 007a
Openhab Device: 8001:8001

(battery device)
----------

Merten 5052xx
https://www.merten.de/uploads/tx_seqdownload/V5051_581_01_web.pdf
http://www.pepper1.net/zwavedb/device/69
http://www.pepper1.net/zwavedb/device/69/69-007a-0001-0004-02-02-1b-01-08.xml

(battery device)
----------

Merten 507001
i found not much more reliable information for this. But with Openhab1 it worked to assign it like one of those 507501, 507502, 507601
https://www.merten.de/uploads/tx_seqdownload/V5075_581_00_web.pdf
http://www.pepper1.net/zwavedb/device/71

maybe there is a way just to give the 507001 the same configuration like the 507501?

cdjackson, 2016-11-14:
Please can you provide the XML file that oh generates for these devices.

Arsur, 2016-11-14:
i dont know why but Oh has created only one XML. But in Habmin /Paperui it shows me all three nodes.
i think this is the Merten 5052xx (not sure)

[node4.zip](https://github.com/openhab/org.openhab.binding.zwave/files/589685/node4.zip)

`<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
  </deviceClass>
  <homeId>0xf74844be</homeId>
  <nodeId>4</nodeId>
  <version>2</version>
  <manufacturer>0x7a</manufacturer>
  <deviceId>0x8001</deviceId>
  <deviceType>0x8001</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>false</beaming>
  <maxBaudRate>9600</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>SWITCH_BINARY</commandClass>
    <commandClass>SWITCH_TOGGLE_BINARY</commandClass>
    <commandClass>SWITCH_ALL</commandClass>
    <commandClass>PROTECTION</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>BASIC</commandClass>
      <basicCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </basicCommandClass>
    </entry>
    <entry>
      <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <libraryType>LIB_SLAVE_ROUTING</libraryType>
        <protocolVersion>2.27</protocolVersion>
        <applicationVersion>2.2</applicationVersion>
      </versionCommandClass>
    </entry>
    <entry>
      <commandClass>CONFIGURATION</commandClass>
      <configurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <configParameters/>
      </configurationCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_ALL</commandClass>
      <switchAllCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
        <mode>SWITCH_ALL_INCLUDE_ON_OFF</mode>
      </switchAllCommandClass>
    </entry>
    <entry>
      <commandClass>PROTECTION</commandClass>
      <protectionCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <localModes>
          <localProtection>UNPROTECTED</localProtection>
          <localProtection>PROTECTED</localProtection>
        </localModes>
        <rfModes>
          <rfProtection>UNPROTECTED</rfProtection>
          <rfProtection>NORFCONTROL</rfProtection>
        </rfModes>
      </protectionCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_TOGGLE_BINARY</commandClass>
      <binaryToggleSwitchCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </binaryToggleSwitchCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_BINARY</commandClass>
      <binarySwitchCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </binarySwitchCommandClass>
    </entry>
    <entry>
      <commandClass>MANUFACTURER_SPECIFIC</commandClass>
      <manufacturerSpecificCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <initSerialNumber>false</initSerialNumber>
        <deviceManufacturer>122</deviceManufacturer>
        <deviceType>32769</deviceType>
        <deviceId>32769</deviceId>
      </manufacturerSpecificCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION</commandClass>
      <associationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <maxGroups>1</maxGroups>
      </associationCommandClass>
    </entry>
  </supportedCommandClasses>
  <securedCommandClasses/>
  <associationGroups>
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors>
    <int>1</int>
  </nodeNeighbors>
  <lastSent>2016-11-14 13:21:20.388 UTC</lastSent>
  <lastReceived>2016-11-14 13:21:20.490 UTC</lastReceived>
</node>`

cdjackson, 2016-11-16:
OH works fine with battery devices, but some Merten devices don't support wakeup class which makes them very (very!) difficult to manage as we don't know when they are awake, and most of the time they are NOT awake.

I've added the 507001 [here](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/462#documents) but it will need some more work to add configuration definitions - it would be great if you can add these. If you don't have an account on the database site then please register and let me know and I'll give you access.

Arsur, 2016-11-21:
first, thank you for your efford.

But how does Merten handle this switches. This two devices are simple wall push buttons. If i activate paaring mode on the 507001 the bottons pair without a problem. It the zave binding not listening the hole time if there is an incoming message (i have realy no plan how zwave communicates in detail, beside that its an mesh network)

cdjackson, 2016-11-21:
> But how does Merten handle this buttons.

I don't really understand what you mean. These are simple wall switches so it just sends out a message when the state changes I assume.

> Does the zave binding not listening the hole time if there is an incoming message

Yes - assuming you have it on all the time.

Arsur, 2016-11-21:
mhm as i understand the problem is just that this switches have no wakeup. So its not discoverable if they are still in the network? Can we still make them work?

cdjackson, 2016-11-21:
They can only send data - this will work fine if they send data to the controller (using for example associations). The problem is that the controller can't send data to the device. 

Maybe I don't understand what you mean.

Arsur, 2016-11-21:
I just ask my self what i have to do to make them work XD

cdjackson, 2016-11-21:
The problem with these switches is likely configuration since it's not possible to know when they are awake - or at least I don't know how to know this since they don't use standards.

Arsur, 2016-11-22:
Iam german and merten seems to  be a company from austria, i will just ask them if they can give me something i can use.

Arsur, 2016-11-22:
what is with this database?
[](http://www.pepper1.net/zwavedb/device/68)
for me it seems to have all the information needed?

cdjackson, 2016-11-22:
Ok - it just needs to be added to the OH database then....

Arsur, 2016-11-28:
okay i put one off them in how can i test if this works?

ps how can i remove false entries? did a little mistake [here](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/57)

cdjackson, 2016-11-28:
What do you want to delete?

Arsur, 2016-11-29:
anything i did here [http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/57](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/57)

The Merten Push Button (505160) is still not working.
I added every Data i could find to this [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/470)

Here is the XMl from openhab [node3.zip](https://github.com/openhab/org.openhab.binding.zwave/files/618659/node3.zip)

Do i have to reset my devices so openhab can find them now? Or maybe i did something wrong with the database entries?

cdjackson, 2016-11-29:
I don't b leave that the binding has been updated with these changes yet.

Arsur, 2016-12-06:
i updated my openhab today, but it still not seems to be working

cdjackson, 2016-12-06:
Which device are we talking about? 507001 or another?

Arsur, 2016-12-06:
no the 505160 (and the 505260). This Battery Push Buttons. I added a lot of info to the Database for the 505160.

cdjackson, 2016-12-06:
I don’t think these devices were completed in the database were they?

Arsur, 2016-12-07:
Is there any data missing?
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/470

cdjackson, 2016-12-07:
> Is there any data missing?

Yes. As it says on the page -:

[image]

cdjackson, 2017-01-31:
Closing due to lack of activity. Please feel free to reopen if there's progress.

closed 2017-01-31 by cdjackson