openhab/org.openhab.binding.zwave #224: GWPN6 Multi-socket PowerNode shows messed power usage

rapiertg, 2016-11-18:
All power usage is shown on meter_watts1. For example with all swiches off if I turn switch_binary6 then watt used are shown as meter_watts1. Loggs below:

2016-11-18 17:59:56.289 [ItemCommandEvent          ] - Item 'hydra1_switch' received command OFF
2016-11-18 17:59:56.292 [ItemStateChangedEvent     ] - hydra1_switch changed from ON to OFF
2016-11-18 18:00:01.790 [ItemStateChangedEvent     ] - hydra1_watts changed from 0.7 to 0
2016-11-18 18:00:01.791 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node15_meter_watts1 changed from 0.7 to 0
2016-11-18 18:00:05.960 [ItemCommandEvent          ] - Item 'hydra2_switch' received command ON
2016-11-18 18:00:05.963 [ItemStateChangedEvent     ] - hydra2_switch changed from OFF to ON
2016-11-18 18:00:12.293 [ItemStateChangedEvent     ] - hydra1_watts changed from 0 to 65.5
2016-11-18 18:00:12.294 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node15_meter_watts1 changed from 0 to 65.5
2016-11-18 18:00:15.165 [ItemCommandEvent          ] - Item 'hydra2_switch' received command OFF
2016-11-18 18:00:15.167 [ItemStateChangedEvent     ] - hydra2_switch changed from ON to OFF
2016-11-18 18:00:21.291 [ItemStateChangedEvent     ] - hydra1_watts changed from 65.5 to 0
2016-11-18 18:00:21.294 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node15_meter_watts1 changed from 65.5 to 0
2016-11-18 18:00:27.636 [ItemCommandEvent          ] - Item 'hydra6_switch' received command ON
2016-11-18 18:00:27.639 [ItemStateChangedEvent     ] - hydra6_switch changed from OFF to ON
2016-11-18 18:00:36.290 [ItemStateChangedEvent     ] - hydra1_watts changed from 0 to 51.2
2016-11-18 18:00:36.296 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node15_meter_watts1 changed from 0 to 51.2
2016-11-18 18:00:40.081 [ItemCommandEvent          ] - Item 'hydra6_switch' received command OFF
2016-11-18 18:00:40.085 [ItemStateChangedEvent     ] - hydra6_switch changed from ON to OFF
2016-11-18 18:00:45.281 [ItemStateChangedEvent     ] - hydra1_watts changed from 51.2 to 0
2016-11-18 18:00:45.281 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node15_meter_watts1 changed from 51.2 to 0

This happens on 2.0.0~20161118032855 deb package. This was the case probably on two previous packages as well.

meter_watts which should agregate all outlets stays at 0 all the time. And meter_watts1 shows last update from all the outlets (but dont sum them).

cdjackson, 2016-11-18:
Please provide a log showing the issue as requested when opening an issue.
If no data is received showing this channel, then please check the configuration of the device.

rapiertg, 2016-11-18:
I enabled DEBUG and turned on switch2 then turned it off after power watts showed on meter_watts1. Then done the same with switch3. Log file is attached.
[zwave.txt](https://github.com/openhab/org.openhab.binding.zwave/files/600564/zwave.txt)

In items I have:
```
Switch  hydra_switch "Hydra all"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary" }
Number  hydra_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts" }
Number  hydra_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh" }
Switch  hydra1_switch "Hydra1"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary1" }
Number  hydra1_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts1" }
Number  hydra1_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh1" }
Switch  hydra2_switch "Hydra2"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary2" }
Number  hydra2_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts2" }
Number  hydra2_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh2" }
Switch  hydra3_switch "Hydra3"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary3" }
Number  hydra3_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts3" }
Number  hydra3_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh3" }
Switch  hydra4_switch "Hydra4"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary4" }
Number  hydra4_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts4" }
Number  hydra4_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh4" }
Switch  hydra5_switch "Hydra5"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary5" }
Number  hydra5_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts5" }
Number  hydra5_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh5" }
Switch  hydra6_switch "Hydra6"     <energy>        { channel="zwave:device:15731ac917a:node15:switch_binary6" }
Number  hydra6_watts "Watt"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_watts6" }
Number  hydra6_kwh "kWh"     <energy>        { channel="zwave:device:15731ac917a:node15:meter_kwh6" }
```
Sitemap:
```
                Frame label="Hydra" {
                        Text item=hydra_watts
                        Group label="Szczegóły"   {
                                Frame label="Całkowite" {
                                        Switch item=hydra_switch
                                        Text item=hydra_watts
                                        Text item=hydra_kwh
                                }
                                Frame label="Hydra1" {
                                        Switch item=hydra1_switch
                                        Text item=hydra1_watts
                                        Text item=hydra1_kwh
                                }
                                Frame label="Hydra2" {
                                        Switch item=hydra2_switch
                                        Text item=hydra2_watts
                                        Text item=hydra2_kwh
                                }
                                Frame label="Hydra3" {
                                        Switch item=hydra3_switch
                                        Text item=hydra3_watts
                                        Text item=hydra3_kwh
                                }
                                Frame label="Hydra4" {
                                        Switch item=hydra4_switch
                                        Text item=hydra4_watts
                                        Text item=hydra4_kwh
                                }
                                Frame label="Hydra5" {
                                        Switch item=hydra5_switch
                                        Text item=hydra5_watts
                                        Text item=hydra5_kwh
                                }
                                Frame label="Hydra6" {
                                        Switch item=hydra6_switch
                                        Text item=hydra6_watts
                                        Text item=hydra6_kwh
                                }
                        }
```

cdjackson, 2016-11-18:
The device doesn't seem to be reporting on this command class - it seems to instead report on the multilevel sensor.

Can you provide the XML please.

rapiertg, 2016-11-18:
xml file attached. This is weird. This device was working fine under latest beta of OH2. At least the main power watts that aggregated all of outlets was working fine and I didn't notice any unussual behavior.

[node15-xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/600826/node15-xml.txt)

cdjackson, 2016-11-18:
I might have got mixed up with one of the other nodes - it looks from the log like it’s simply not sending the root endpoint data at all. I’m not sure what else to suggest on this - you could try polling to see if it returns data, but otherwise if the node doesn’t send anything for this channel, then the binding can’t do anything...

rapiertg, 2016-11-20:
Sorry, could not find it anywhere. What is the way of forcing it to use pulling in OH2? The thing was added with Paper UI.

cdjackson, 2016-11-20:
Polling is configured in the thing configuration - you can set a time which defaults to 1800 seconds, but can be set a lot lower (down to around 15 seconds I think)

rapiertg, 2016-11-20:
I got it to resfresh. Read somwhere that it needs to be pooled. This is working.

What is still not working is overall power usage. 
channel="zwave:device:15731ac917a:node15:meter_watts"
and
channel="zwave:device:15731ac917a:node15:meter_watts1"
seems to be the same logical meter. The same is with switch. If i turn switch_binary1 on I get:

> 2016-11-20 18:43:37.420 [ItemCommandEvent          ] - Item 'hydra1_switch' received command ON
> 2016-11-20 18:43:37.423 [ItemStateChangedEvent     ] - hydra1_switch changed from OFF to ON
> 2016-11-20 18:43:40.998 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node16_meter_watts1 changed from 0 to 9.6
> 2016-11-20 18:43:41.000 [ItemStateChangedEvent     ] - hydra1_watts changed from 0 to 9.6
> 2016-11-20 18:43:43.945 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node16_switch_binary changed from OFF to ON
> 2016-11-20 18:43:43.947 [ItemStateChangedEvent     ] - hydra_switch changed from OFF to ON
> 2016-11-20 18:43:44.102 [ItemStateChangedEvent     ] - zwave_device_15731ac917a_node16_meter_watts changed from 0 to 9.6
> 2016-11-20 18:43:44.104 [ItemStateChangedEvent     ] - hydra_watts changed from 0 to 9.6

The overall meter and overall switch was working fine with OH1.X.

rapiertg, 2016-11-23:
Just tried installing OH1.x as I found my configs in the backup. It is still broken, so probably the hardware is broken, so this can be closed.

closed 2016-11-23 by rapiertg