openhab/org.openhab.binding.zwave #225: iModuleDouble 2nd Channel Not Working

carryonrewardless, 2016-11-19:
Ref
* OH2 Version: Version 2
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.0.0.201611181914

If the problem is associated with a device, please provide the following -:

* Device name: iModuleDouble
* Device version: 1122:5501
* Link to [database] : http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/348
*  XML file attached
Debug log file extract attached.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.
[node6.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/601217/node6.xml.txt)
[node6.1911A.txt](https://github.com/openhab/org.openhab.binding.zwave/files/601219/node6.1911A.txt)

The second channel is not functioning. Using 
Command received zwave:device:71cd1c22:node6:switch_binary1 --> ON   
or 
Command received zwave:device:71cd1c22:node6:switch_binary2 --> ON
results in no response whereas 
Command received zwave:device:71cd1c22:node6:switch_binary --> ON
brings both channels to ON status as seen in the logfile.
Hardwiring is correct as wall switches correctly operate the two lights.

The previous issue to set this device up can be found here: https://github.com/openhab/openhab2-addons/issues/791

cdjackson, 2016-11-19:
Please provide a full debug log. You’ve filtered the log to only show node 6, and this removes all received data responses from the log so it’s of no use - sorry.

carryonrewardless, 2016-11-19:
Ok try again. I was trying to keep the file size down.
Thanks
[OpenhabLogExtract.DOCX](https://github.com/openhab/org.openhab.binding.zwave/files/601403/OpenhabLogExtract.DOCX)

carryonrewardless, 2016-11-19:
The DOCX file is a .txt file that has been zipped and then renamed to DOCX extension to pass the upload gate.

cdjackson, 2016-11-19:
Thanks. This is useful…

cdjackson, 2016-11-20:
I've created a (hopefully!) fix for this  which should be available in an hour or so... Let me know how it goes...

carryonrewardless, 2016-11-20:
Early testing suggests you have nailed it! Well done Chris. Big thanks!

cdjackson, 2016-11-20:
Thanks - I'll close this then - any issues, reopen or open a new one :)

closed 2016-11-20 by cdjackson