openhab/org.openhab.binding.zwave #228: Aeon Smart Switch Gen5 EU is identified as ZW075 Smart Energy Switch 3rd Edition #4798 and is not working

sennerholm, 2016-11-20:
I have some Aeon Smart Switch Gen5 EU, and they are identified as ZW075 Smart Energy Switch 3rd Edition and not working (trying to toogle power switch). In Razberry z-way they are correctly identified and it's possible to switch power.
I have tried to exclude and rejoin but it's not working. It's the same for a all four devices I have. I think it's because it's product is ZW075-C16 and the regexp rule for "ZW075 Smart Energy Switch 3rd Edition" says anything starting with ZW075 (link to db is: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/83).

It's also showing up as having a battery in Openhab2 but not in razberry, but jackson page above link to the correct manual, but the device has the wrong name on the page and maybe wrong config information wish make the device not working. 

I attach an xml file for one of the device (from: /var/lib/openhab2/zwave/node7.xml).

I have found the device in the links from Razberrys page:
http://pepper1.net/zwavedb/device/780
Where it mentions:
Product Code:	ZW075-C16 and

More links:
http://pepper1.net/zwavedb/device/780/780-0086-0003-004b-00-03-53-02-16.xml
http://products.z-wavealliance.org/products/1045

Expected Behavior

That it's added as
Aeon Smart Switch Gen5 [EU]
and it's possible to toggle the power.

Current Behavior

It's added as:
ZW075 Smart Energy Switch 3rd Edition
an it's not possible to toogle power.

Some debug log:

 I runned 
openhab> log:set DEBUG org.openhab.binding.zwave

And then tried to turn the switch on:
2016-11-20 13:03:55.045 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Command received zwave:device:378edfea:node7:switch_binary --> ON
2016-11-20 13:03:55.046 [DEBUG] [dclass.ZWaveBinarySwitchCommandClass] - NODE 7: Creating new message for application command SWITCH_BINARY_SET
2016-11-20 13:03:55.049 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
2016-11-20 13:03:55.051 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 1. Queue={}
2016-11-20 13:03:55.053 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0A 00 13 07 03 25 01 FF 25 99 85
2016-11-20 13:03:55.056 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 7: Sending REQUEST Message = 01 0A 00 13 07 03 25 01 FF 25 99 85
2016-11-20 13:03:55.080 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
2016-11-20 13:03:55.084 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-11-20 13:03:55.086 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8
2016-11-20 13:03:55.089 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8
2016-11-20 13:03:55.091 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Response[0x01], priority=High, dest=255, callback=0, payload=01
2016-11-20 13:03:55.093 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 7: Sent Data successfully placed on stack.
2016-11-20 13:03:55.205 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 99 00 00 0E 7C
2016-11-20 13:03:55.212 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-11-20 13:03:55.215 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 99 00 00 0E 00 00 72
2016-11-20 13:03:55.221 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 13 99 00 00 0E 00 00 72
2016-11-20 13:03:55.226 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=99 00 00 0E
2016-11-20 13:03:55.230 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 7: SendData Request. CallBack ID = 153, Status = Transmission complete and ACK received(0)
2016-11-20 13:03:55.234 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 7: Starting initialisation from DONE
2016-11-20 13:03:55.237 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@122d3be already registered
2016-11-20 13:03:55.240 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Set, dest=7, callback=153, payload=07 03 25 01 FF
2016-11-20 13:03:55.243 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=99 00 00 0E
2016-11-20 13:03:55.245 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=SendData, callback id=153, expected=SendData, cancelled=false transaction complete!
2016-11-20 13:03:55.248 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
2016-11-20 13:03:55.250 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2016-11-20 13:03:55.252 [DEBUG] [curityCommandClassWithInitialization] - NODE 7: updating lastSentMessageTimestamp
2016-11-20 13:03:55.255 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 7: Response processed after 185ms/4403ms.

Nothing happend. Then I try to turn it off:

2016-11-20 13:04:06.541 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Command received zwave:device:378edfea:node7:switch_binary --> OFF
2016-11-20 13:04:06.543 [DEBUG] [dclass.ZWaveBinarySwitchCommandClass] - NODE 7: Creating new message for application command SWITCH_BINARY_SET
2016-11-20 13:04:06.545 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
2016-11-20 13:04:06.549 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0A 00 13 07 03 25 01 00 25 9A 79
2016-11-20 13:04:06.552 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 7: Sending REQUEST Message = 01 0A 00 13 07 03 25 01 00 25 9A 79
2016-11-20 13:04:06.554 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 0. Queue={}
2016-11-20 13:04:06.562 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
2016-11-20 13:04:06.569 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-11-20 13:04:06.571 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8
2016-11-20 13:04:06.573 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8
2016-11-20 13:04:06.575 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Response[0x01], priority=High, dest=255, callback=0, payload=01
2016-11-20 13:04:06.576 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 7: Sent Data successfully placed on stack.
2016-11-20 13:04:06.741 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 9A 00 00 0E 7F
2016-11-20 13:04:06.757 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2016-11-20 13:04:06.760 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 9A 00 00 0E 00 00 71
2016-11-20 13:04:06.762 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 13 9A 00 00 0E 00 00 71
2016-11-20 13:04:06.765 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=9A 00 00 0E
2016-11-20 13:04:06.766 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 7: SendData Request. CallBack ID = 154, Status = Transmission complete and ACK received(0)
2016-11-20 13:04:06.768 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 7: Starting initialisation from DONE
2016-11-20 13:04:06.770 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@122d3be already registered
2016-11-20 13:04:06.772 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Set, dest=7, callback=154, payload=07 03 25 01 00
[... cut ...]
2016-11-20 13:04:06.776 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
2016-11-20 13:04:06.777 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2016-11-20 13:04:06.779 [DEBUG] [curityCommandClassWithInitialization] - NODE 7: updating lastSentMessageTimestamp
2016-11-20 13:04:06.780 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 7: Response processed after 211ms/4403ms.

Sincerely
Mikael

sennerholm, 2016-11-20:
Hi @cdjackson

Created an issue in the rigth place now hopefully.

Sincerely
Mikael

cdjackson, 2016-11-20:
I don't really see any problem at the moment.

1) The device is identified ok
2) from the log, it seems to work.

[image]

I guess it's not really working is what you're saying, but clearly the device is responding and the log looks ok, so I'm not really sure what I can do.

sennerholm, 2016-11-20:
I think the device get in some kind of "limbo" state when it's added and
have a battery according to the "Smart Energy Switch 3rd " definition. Is
it a easy way for me to remove that?

cdjackson, 2016-11-20:
Sorry -  don’t know what you mean? Is there an easy way to remove what exactly?

Can you supply the XML for this device from your OH2 system?

sennerholm, 2016-11-20:
Oups, forgot to add it to the issue on OH2. I add it now. 
[node7.xml.gz](https://github.com/openhab/org.openhab.binding.zwave/files/602085/node7.xml.gz)

In the z-way server (where it works) it doesnt show any binary sensor or battery level so my idea is that openhab2 when try to get that data in some way confuse the device and I want to try with a cleaner specification of the device. 

Sincerely
Mikael

cdjackson, 2016-11-20:
Ah - ok, not sure why these are there. I will delete them, but I don’t think it will help. From the log you sent earlier, the commands were being sent ok…

Let’s see - I’ll update the database later with this.

sennerholm, 2016-11-20:
Strange, when I try to read another device, it works to power toggle it if I first toggled the "Binary Sensor" wish doesn't exists. And afterwards it was working independently of the binary sensor, but the readout for power consumption doesn't work. The configparameter I can set on the device is totaly different between the z-way (I think more true parameters) than what I can set in openhab2.

I try to exclude/include another of my devices and see what happens.

sennerholm, 2016-11-20:
I tried to exclude/include another of the devices, and it works to toogle it. Strange, I tried the same yesterday but then it doesn't work. Now I'm in debug logging mode, but that this change the behavior? Or I'm just lucky?

cdjackson, 2016-11-20:
I just did a quick comparison of the parameters in the manual and the database, and they look very similar. The only one that looked different was parameter 20. Maybe this is a version issue. My check was very quick as well, so maybe there’s more, but it certainly doesn’t look ‘totally different’.

If you could take a closer look to check differences from the database and your manual it would be useful.

cdjackson, 2016-11-20:
No - debug mode doesn’t change anything at all - it just changes the lower level logging, but in the binding, it’s exactly the same.

sennerholm, 2016-11-20:
I compared to z-way. The user manual is the same as attached to http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/83. I haven't got the advanced part.

cdjackson, 2016-11-20:
Ok - if you don’t mind it would be good to compare with the database to see what the differences are so they can be fixed. If can always do this, but it might take a lot of time ;)

sennerholm, 2016-11-20:
If you give me some directions I will try (my first time a device isn't recognised correctly in openhab)

cdjackson, 2016-11-20:
Thanks - if you can just look down through the list on the device database page to see what is different and make a note. If it just needs some changes, I can give you edit access, but if something needs deleting then I will need to do it...

cdjackson, 2017-01-03:
Closing due to no activity. Please feel free to update the database if you still want to use this device.

closed 2017-01-03 by cdjackson