openhab/org.openhab.binding.zwave #229: Enerwave ZWN-RSM2 change device ID in last batch, not recognized in database

luiscampiz, 2016-11-20:
Hello, I thing Enerwave change the deviceID in the new batch of this device. I guess they make a mistake in the first batch and put the deviceID of another device and now they corrected it. For this reason the new batch is not recognised on discovery.
I have not found a documentation for the new database, and how to modified without a PR.
I hope this can be resolved soon, excelent work to all of the contributors.
Thanks. (sorry for my english)
* OH2 Version: SNAPSHOT 2.0.0.201611181914
* Binding Version: SNAPSHOT 2.0.0.201611181914
* Device name: Enerwave ZWN-RSM2
* Device version: In node2.xml: "v4"   and  in Habmin|Properties|Firmware Version: "v1.0" 
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): It is no in that link but i found it here: [github database](https://github.com/openhab/openhab/blob/master/bundles/binding/org.openhab.binding.zwave/database/enerwave/zwn-rsm2.xml)
* XML just in case:
`<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
  </deviceClass>
  <homeId>0xdf210ab8</homeId>
  <nodeId>2</nodeId>
  <version>4</version>
  <manufacturer>0x11a</manufacturer>
  <deviceId>0x5606</deviceId>
  <deviceType>0x101</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>VERSION</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>MULTI_INSTANCE</commandClass>
    <commandClass>MULTI_INSTANCE_ASSOCIATION</commandClass>
    <commandClass>SWITCH_BINARY</commandClass>
    <commandClass>BASIC</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>SWITCH_ALL</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>BASIC</commandClass>
      <basicCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </basicCommandClass>
    </entry>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>CONFIGURATION</commandClass>
      <configurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <configParameters/>
      </configurationCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION</commandClass>
      <associationCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <maxGroups>2</maxGroups>
      </associationCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_ALL</commandClass>
      <switchAllCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
        <mode>SWITCH_ALL_INCLUDE_ON_OFF</mode>
      </switchAllCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_BINARY</commandClass>
      <binarySwitchCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </binarySwitchCommandClass>
    </entry>
    <entry>
      <commandClass>MANUFACTURER_SPECIFIC</commandClass>
      <manufacturerSpecificCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <initSerialNumber>false</initSerialNumber>
        <deviceManufacturer>282</deviceManufacturer>
        <deviceType>257</deviceType>
        <deviceId>22022</deviceId>
      </manufacturerSpecificCommandClass>
    </entry>
    <entry>
      <commandClass>MULTI_INSTANCE</commandClass>
      <multiInstanceCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>3</versionSupported>
        <endpoints>
          <entry>
            <int>1</int>
            <endPoint>
              <deviceClass>
                <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
                <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
                <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
              </deviceClass>
              <endpointId>1</endpointId>
              <supportedCommandClasses>
                <entry>
                  <commandClass>BASIC</commandClass>
                  <basicCommandClass>
                    <endpoint reference="../../../.."/>
                    <version>0</version>
                    <instances>0</instances>
                    <versionSupported>0</versionSupported>
                    <isGetSupported>true</isGetSupported>
                  </basicCommandClass>
                </entry>
                <entry>
                  <commandClass>SWITCH_BINARY</commandClass>
                  <binarySwitchCommandClass>
                    <endpoint reference="../../../.."/>
                    <version>1</version>
                    <instances>1</instances>
                    <versionSupported>1</versionSupported>
                    <isGetSupported>true</isGetSupported>
                  </binarySwitchCommandClass>
                </entry>
              </supportedCommandClasses>
            </endPoint>
          </entry>
          <entry>
            <int>2</int>
            <endPoint>
              <deviceClass>
                <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
                <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
                <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
              </deviceClass>
              <endpointId>2</endpointId>
              <supportedCommandClasses>
                <entry>
                  <commandClass>BASIC</commandClass>
                  <basicCommandClass>
                    <endpoint reference="../../../.."/>
                    <version>0</version>
                    <instances>0</instances>
                    <versionSupported>0</versionSupported>
                    <isGetSupported>true</isGetSupported>
                  </basicCommandClass>
                </entry>
                <entry>
                  <commandClass>SWITCH_BINARY</commandClass>
                  <binarySwitchCommandClass>
                    <endpoint reference="../../../.."/>
                    <version>1</version>
                    <instances>1</instances>
                    <versionSupported>1</versionSupported>
                    <isGetSupported>true</isGetSupported>
                  </binarySwitchCommandClass>
                </entry>
              </supportedCommandClasses>
            </endPoint>
          </entry>
        </endpoints>
        <useDestEndpointAsSource>false</useDestEndpointAsSource>
        <endpointsAreTheSameDeviceClass>true</endpointsAreTheSameDeviceClass>
      </multiInstanceCommandClass>
    </entry>
    <entry>
      <commandClass>MULTI_INSTANCE_ASSOCIATION</commandClass>
      <multiAssociationCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <maxGroups>2</maxGroups>
      </multiAssociationCommandClass>
    </entry>
    <entry>
      <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <libraryType>LIB_SLAVE_ROUTING</libraryType>
        <protocolVersion>4.5</protocolVersion>
        <applicationVersion>1.0</applicationVersion>
      </versionCommandClass>
    </entry>
  </supportedCommandClasses>
  <securedCommandClasses/>
  <associationGroups>
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <associations/>
      </associationGroup>
    </entry>
    <entry>
      <int>2</int>
      <associationGroup>
        <index>2</index>
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors>
    <int>1</int>
  </nodeNeighbors>
  <lastSent>2016-11-20 04:02:47.423 UTC</lastSent>
  <lastReceived>2016-11-20 04:02:47.477 UTC</lastReceived>
</node>`

BrandonArp, 2016-11-22:
I would also love to see this device included.  I can help test as well.

cdjackson, 2016-11-22:
If it's really not in the database, then please feel free to add it at www.cd-jackson.com. I am a bit surprised that its in the OH1 database, and not the OH2 since pretty much all devices were copied across, but I also don't immediately see it there...

luiscampiz, 2016-11-22:
@cdjackson . Hi, It is in the github database but with incorrect deviceID. I detalles the exact problem in the first post.
It is not in your site database. Actualy it is not even the Manufacturer.

cdjackson, 2016-11-22:
Ok - I guess you don't mean that the deviceId is incorrect, but just different for your device... For the database, I'm sure the manufacturer is in there since every manufacturer is included (as defined by ZWave in their official doc). My guess is that it has a different name - can you check the ID is 0x11a.

luiscampiz, 2016-11-22:
Yes it is different DeviceID. I can confirm that 0x11a is the ManufacturerID. In this issue [#2595](https://github.com/openhab/openhab/issues/2595) somebody says it has the same DeviceID as another product from the same Manufacturer. But that was another batch of the device I guess (firmware error?). Thats the reason I say the deviceID was incorrect in the earlier product, and I can see that what they do was add it to the openhab 1 database but not in the new one. I have already an account in your site but I dont have write access, maybe I can help. I just dont know how not to affect the devices with the older ID. Sorry for the confusion and my English, thank you very much for your effort.

cdjackson, 2016-11-22:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/465

closed 2016-11-22 by cdjackson

njcrawford, 2016-12-27:
I'm having trouble getting OH1 to recognize this device. I've downloaded and installed the latest Zwave binding from CloudBees, version 1.9.0-201612250211. Is there something more I need to do? Or perhaps it is only supported in OH2?

For reference, this is the line in the log that makes me think OH isn't recognizing the device:
2016-12-26 22:58:26.873 [DEBUG] [.z.i.config.ZWaveConfiguration] - NODE 13: No database entry: Enerwave [ID:5606,Type:101]
This line seems to be repeated every 6 seconds or so.