openhab/org.openhab.binding.zwave #230: Everspring door contact HSM02 not working as expected

yellowmm, 2016-11-21:
* OH2 Version: openhab-offline-2.0.0-SNAPSHOT from 20161121
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):
org.openhab.binding.zwave/2.0.0-SNAPSHOT from 20161121

* Device name: Everspring door/window contact HSM02
* Device version: 1.1
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/34

**Main Problem**
First of all this device is working without any issues in OH1 (1.8.3). But when I move to OH2, the device is well recognised, but it is not sending any updates (e.g. OPEN/CLOSE) to the controller.

1) I dont know why, but this device is recognised as Everspring SM103 door/window contact in OH1 & OH2. Seems to be a minor issue because it works in OH1.
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/35

2) But when you have a look at the device Information you will see the following difference which may cause the problem:
[image]
__Snapshot Habmin, OH1__

[image]
__Snapshot Habmin2, OH2__

The generic class and specific class are different.
BINARY_SENSOR vs. ALARM_SENSOR
ROUTING_SENSOR_BINARY vs. SMOKE_SENSOR_ROUTING

cdjackson, 2016-11-21:
Is there a logfile available?

yellowmm, 2016-11-21:
Sure, have a look at NODE 2. 
The both messages OPEN/CLOSE are available where I opened and closed the contact. But Node view tells 100% timeout.

[zwave_log_node2_everspring_hsm02.zip](https://github.com/openhab/org.openhab.binding.zwave/files/604550/zwave_log_node2_everspring_hsm02.zip)

cdjackson, 2016-11-21:
From the log, it looks like the device isn’t receiving messages from the controller - the controller is receiving the device, but not the other way around it seems.

I can’t tell from the log if the status is processed correctly - I guess not from your report, so this might also be a separate issue - possibly with the database.

yellowmm, 2016-11-21:
This is very strange. As I said above, OH1 works perfectly.
Does it mean that e.g. 
`NODE 2: Updating channel state zwave:device:1586da72353:node2:sensor_binary to ON [OnOffType]`
the related item is updated only in case the according node is sending the correponding ACK?

This is from events.log:
`2016-11-21 19:50:39.920 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from ONLINE: Node initialising: FAILED_CHECK to ONLINE: Node initialising: WAIT
2016-11-21 19:55:08.928 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from ONLINE: Node initialising: WAIT to ONLINE
2016-11-21 19:55:21.038 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller
`

cdjackson, 2016-11-21:
> NODE 2: Updating channel state zwave:device:1586da72353:node2:sensor_binary to ON [OnOffType]
> the related item is updated only in case the according node is sending the correponding ACK?
> 
No - this is what I meant by there may also be a problem with the database as well since this should update the channel.

yellowmm, 2016-11-21:
Ok. And now???

cdjackson, 2016-11-22:
> Ok. And now???

It needs to be looked at to see what's wrong...

yellowmm, 2016-11-22:
This is the only Thing I can see from the logs during active changes from the _contact_ (OPEN/CLOSE)

```
2016-11-21 20:27:45.471 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 255
2016-11-21 20:27:45.528 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 99
2016-11-21 20:27:47.969 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 0
2016-11-21 20:27:48.024 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 0
2016-11-21 20:27:45.461 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from ONLINE: Node initialising: WAIT to ONLINE
2016-11-21 20:27:55.539 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from ONLINE to OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller

```
```
2016-11-21 20:56:30.492 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 255
2016-11-21 20:56:30.546 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 99
2016-11-21 20:56:36.135 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_BINARY, value = 0
2016-11-21 20:56:36.262 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got a value event from Z-Wave network, endpoint = 0, command class = BASIC, value = 0
2016-11-21 20:56:30.482 [hingStatusInfoChangedEvent] - 'zwave:device:1586da72353:node2' changed from OFFLINE (COMMUNICATION_ERROR): Node is not communicating with controller to ONLINE
```
_hingStatusInfoChangedEvent_ is from events.log

I think that an item is only being update when the following TRACE appears (saw this on all other nodes):

`zwave_device_1586da72353_node2:sensor_binary changed from ON to OFF`

But this was never seen.

cdjackson, 2016-11-27:
> I think that an item is only being update when the following TRACE appears (saw this on all other nodes):
>
> zwave_device_1586da72353_node2:sensor_binary changed from ON to OFF
>
> But this was never seen.

In the log, I see this -:

```
2016-11-21 19:55:08.987 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Updating channel state zwave:device:1586da72353:node2:sensor_binary to ON [OnOffType]
2016-11-21 19:55:09.032 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Updating channel state zwave:device:1586da72353:node2:sensor_binary to ON [OnOffType]
2016-11-21 19:55:12.902 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Updating channel state zwave:device:1586da72353:node2:sensor_binary to OFF [OnOffType]
```

It appears to be working to me...  Maybe you have an issue with the item type, or item linking, or something in the item domain...

yellowmm, 2016-12-30:
After getting more and more familiar with OH2 I found the reason why the door contact is not working as expected.
In OH1 my item-config is as follows:
`Contact frontDoor "myDoor" <door> { channel="zwave:device:1586da72353:node2:sensor_binary" }
`
But when you have a look at the log above you can see that this item is an **OnOffType**. Thats why I have tried to configure the conact as a switch as follows:
`Switch frontDoor "myDoor" <door> { channel="zwave:device:1586da72353:node2:sensor_binary" }
`
This is working now. 
But from my point of view, a contact is not a switch. Why must I configure it as a **Switch**?

yellowmm, 2016-12-30:
Update: I saw you introduced a new type [sensor_door](https://community.openhab.org/t/openhab-2-z-wave-channel-types/8462/16).

I have updated the zwave database accordingly.

closed 2017-01-03 by yellowmm