openhab/org.openhab.binding.zwave #231: Everspring SF812 smoke detector has wrong alarm trigger

yellowmm, 2016-11-21:
* OH2 Version: openhab-offline-2.0.0-SNAPSHOT from 20161121
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):
org.openhab.binding.zwave/2.0.0-SNAPSHOT from 20161121

* Device name: Everspring SF812 smoke detector 
* Device version: n/a
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/33

I think the __Endpoint__ configuration for this device is not correct. As you can see in the database, ALARM and SENSOR_ALARM have the same type: _Alarm (Smoke)_ 

The result is as follows in OH2/Habmin2:
[image]

When you have a look to the documentation, the ALARM is once used to
- Trigger tampering (Command Class Alarm, Alarm Type = 0x01, Alarm Level = 17(0x11))
- Low battery (Command Class Alarm, Alarm Type = 0x01, Alarm Level = 255(0xFF))

Could this been changed according to the device specification? From my point of view these are useful features for a real smart home.

cdjackson, 2016-12-18:
The documentation isn't very clear. I suspect that this uses the notification class - can you provide a debug log please showing these events.

yellowmm, 2016-12-30:
The zwave database for SF812 has been updated on _2016-12-19 13:00:25_
Looks quite better now.

yellowmm, 2017-01-03:
The base functionality is working when you consider to define all the respective items as `Switches`.
Why is `alarm_smoke`, `alarm_tamper` and `alarm_battery` defined as `OnOffType` instead of `OpenClosedType`.
This seem to be a general kind of issue/misunderstanding because I saw this on many other smoke detectors in the zwave-database?
What do you think?

cdjackson, 2017-01-03:
Sorry - what do I think about what exactly? Do you mean that there is a misunderstanding about the types? Currently I think I'm using the types defined in [ESH](https://eclipse.org/smarthome/documentation/development/bindings/thing-definition.html) - eg smoke is an ON/OFF switch. Why do you think an OPEN/CLOSE contact is better.

You need to use the correct type - sorry - this isn't the same as OH1 where you are free to use whatever type you choose. In OH2, the binding has no knowledge about items, so if you don't use the right type in your item, the binding has no way to know what you have set and it will not work.

yellowmm, 2017-01-03:
From my point of view an `OnOffType` is a _RW_ type (e.g. a wall mounted light-switch) which can also be manually altered. In ESH its a `Switch`.
An `OpenCloseType` is a _R_ type which cannot be written. ESH -> `Contact` (e.g. a door_sensor)

Thats why I assume these kind of items required for any alarm_sensors shall be of `Contact` type.

The pitfall of the zwave-XML definintion is that an `OnOffType` can only be combined with a `Switch`. If you doing this by `Contact` the items is ignored in the OH2 database.

The main point is, what do we define as a `Contact` and what do we define as a `Switch`?

cdjackson, 2017-01-03:
> The main question may: What do we define as a Contact and what do we define as a Switch?

There is no point in discussing this here since this is already defined in the link I posted - I am using the types defined by ESH, so if we want to change things, then we should change the ESH definitions.  There have been disucssions about trying to widen this definition, but at the moment nothing has changed and I don't see this changing any time soon.

yellowmm, 2017-01-04:
OK. Understand and accepted...

One minor question for understanding, at the zwave database, [OpenHAB-2 Channel Types](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/openhab-2-channel-types) it's mentioned that for several alarms the following _Option_ is defined (e.g.):

Label:`Smoke alarm`, Name:`alarm_smoke` type:`Switch` Option: `ON = Ok, OFF = Alarm`

Issn't the opposite truth? In case a `Switch` moves to `ON` an alarm is triggered?
That's what I see when I test the device. I am just wondering...

cdjackson, 2017-01-04:
Yes - you are likely right that ON = Alarm.

closed 2017-01-31 by cdjackson