openhab/org.openhab.binding.zwave #234: Secure inclusion fails at 'step SECURITY_NETWORK_KEY_SET'

brstein, 2016-11-23:
Testsetup: OpenHAB2, daily build # 567 (dated Nov., 4th) with Aeotec Z-Stick Gen5 and AeonLabs Multisensor 6

Habmin was set to do a secure inclusion on any device. The sensor was clicked twice on the include button to do a secure include. The inclusion process was logged in the following file, where NODE 7 is the node to be included:

[openhab.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/608571/openhab.log.txt)

The inclusion process fails (at the bottom of the logfile) with 

`2016-11-15 22:04:40.577 [ERROR] [curityCommandClassWithInitialization] - NODE 7: SECURITY_INCLUSION_FAILED Failed at step SECURITY_NETWORK_KEY_SET: 10000ms passed since last request was sent, secure inclusion failed.`

Before this final failure occurs, the following is stated in the log:

`2016-11-15 22:04:29.896 [ERROR] [ssage.ApplicationCommandMessageClass] - NODE 7: Unknown command class 0xffffff98`

I am not sure if I have clumsy fingers or if this is a bug. Any help is appreciated,

   brstein

cdjackson, 2016-11-23:
Security is not currently supported in the OH2 binding. It's being worked on at the moment, but the current binding doesn't have support.

I will close this for now - once the security classes are added (very soon I hope now) we can look at this again if things don't work, but there is no point in keeping this open as the security implementation is a total rewrite from the OH1 implementation.

closed 2016-11-23 by cdjackson

brstein, 2016-11-23:
Thank you for this information.
I'm currently very curious about the security features - Please announce it, if the new security classes are ready for testing.
Regards,
   brstein

cdjackson, 2016-11-23:
Will do. It's not available yet as I'm working on it at the moment but the initial key exchange is working so I hope it will be soon.