openhab/org.openhab.binding.zwave #242: Z-Wave.Me Smoke Detector + Siren

hmerk, 2016-11-27:
Hi @cdjackson 
I just got my hands of two Z-Wave.Me devices (product code 004001 / pepper database 656) that are similar to the Popp 004001 (pepper database 725) smoke detectors.

Comparing the  devices with your entry for Popp 004001, there seems to be an error in your database, causing my devices not to work properly. Switching the siren is not working reliable.

Another strange thing, Paper-UI shows zwave_manufacturer 340 and zwave_devicetype 256

Best
Hans-Jörg

cdjackson, 2016-11-27:
Can you provide some information on what you think the error is - it would be appreciated :)

If you can provide the XML, debug logs, it would be good. Otherwise I’m just guessing….

hmerk, 2016-11-27:
I guess that the channel "switch_binary" is missing command class "BASIC"

      <channel id="switch_binary" typeId="switch_binary">
        <label>Switch</label>
        <properties>
          <property name="binding:*:OnOffType">SWITCH_BINARY</property>
        </properties>
      </channel>

I will produce debug logs tomorrow, will not take the risk to wake up my wife if it eventually works, what it did one time ...

cdjackson, 2016-11-27:
Ok - thanks. That’s very simple to fix - just tick a box on the database ;)

cdjackson, 2016-11-28:
Updated http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/229#endpoints <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/229#endpoints>

hmerk, 2016-12-22:
Devices are working as expected, Thanks @cdjackson !

closed 2016-12-22 by hmerk