openhab/org.openhab.binding.zwave #243: Add Everspring AN180 On/Off Mini Plug to database

hmerk, 2016-11-27:
Please add AN180 zo your database (AN181 with measurement already exists).
openHAB did not create a node?.xml, so I can just provide link to pepper database.

http://www.pepper1.net/zwavedb/device/687

Thanks
Hans-Jörg

cdjackson, 2016-11-28:
We need to get the XML generated - if it’s not creating one, then we should grab a debug log so we can find out why.

hmerk, 2016-11-28:
I will be travelling until next week Wdnesday, so will try to get the needed information when I return.

Best
Hans-Jörg

hmerk, 2016-11-28:
Hi Chris,
found some minutes to try modifying your everspring_an181_0_0.xml.

NEW **everspring_an180_0_0.xml** 

<?xml version="1.0" encoding="UTF-8"?>
<thing:thing-descriptions bindingId="zwave"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns:thing="http://eclipse.org/smarthome/schemas/thing-description/v1.0.0"
  xsi:schemaLocation="http://eclipse.org/smarthome/schemas/thing-description/v1.0.0
                      http://eclipse.org/smarthome/schemas/thing-description/v1.0.0">

  <thing-type id="everspring_an180_00_000" listed="false">
    <label>AN180 Mini Plug Switch (Gen5)</label>
    <description><![CDATA[
Mini Plug Switch (Gen5)<br /><h1>Overview</h1><p>The Plug is designed to control the on/off status appliances load in your house. At 220-240V voltage, this Plug can support up to 2500W resistive. </p> <br /><h2>Inclusion Information</h2><p>Press the link key three times within 1.5 seconds to put the unit into inclusion mode.</p> <br /><h2>Exclusion Information</h2><p>Press the link key three times within 1.5 seconds to put the unit into exclusion mode.</p>
    ]]></description>

    
    <channels>
      <channel id="switch_binary" typeId="switch_binary">
        <label>Switch</label>
        <properties>
          <property name="binding:*:OnOffType">SWITCH_BINARY</property>
        </properties>
      </channel>
    </channels>

    
    <properties>
      <property name="vendor">Everspring</property>
      <property name="modelId">AN180</property>
      <property name="manufacturerId">0060</property>
      <property name="manufacturerRef">0004:0007</property>
      <property name="dbReference">392</property>
      <property name="defaultAssociations">1,2</property>
    </properties>

    
    <config-description>

      
      <parameter-group name="configuration">
        <context>setup</context>
        <label>Configuration Parameters</label>
      </parameter-group>

      <parameter name="config_1_2" type="integer" groupName="configuration"
                 min="1" max="99">
        <label>1: Command value</label>
        <description><![CDATA[
Set Basic Set Command value to be sent to group 2 when switch is turned on<br /><h1>Overview</h1><p>Meaningful range is 0-99 &amp; 255 where</p> <p>0 = No command will be sent<br />1-99 = Turn on switches and dimmers listening to Group 2 to indicated % level (ON for switches)<br />255 = Turn on switches and dimmers (full power) listening to Group 2</p> <p>Command will be sent if the mini plug is turned on via the button or if it receives a remote command.</p> <p>Basic Set Command = 0 (Off) will be sent to Group 2 only if the button is used to turn off the mini plug, not when the mini plug is turned off via remote command.</p>
        ]]></description>
        <default>255</default>
        <options>
          <option value="0">No Basic Set Command will be sent</option>
          <option value="255">Basic Set Command ON will be sent</option>
        </options>
        <limitToOptions>false</limitToOptions>
      </parameter>

      <parameter name="config_2_1" type="integer" groupName="configuration"
                 min="3" max="25">
        <label>2: Delaying time</label>
        <description>Set the delaying time to report to Group 1 in seconds</description>
        <default>3</default>
      </parameter>

      <parameter name="config_3_1" type="integer" groupName="configuration">
        <label>3: Remember status</label>
        <description>Set whether to remember the last status</description>
        <default>1</default>
        <options>
          <option value="0">Do not remember</option>
          <option value="1">Remember</option>
        </options>
      </parameter>

      
      <parameter-group name="association">
        <context>link</context>
        <label>Association Groups</label>
      </parameter-group>

      <parameter name="group_1" type="text" groupName="association">
        <label>1: Target for automatic reports</label>
        <description><![CDATA[
Binary Switch Report, Notification Report, Device Reset Locally Notification, Meter Report<br /><h1>Overview</h1><p>When powered on, the device will send (Notification Type = 0x08, Event = 0x01) to the node of Group 1. When overload occurs, the device will send (Notification Type = 0x08, Event = 0x08) to the node of Group 1.</p>
        ]]></description>
      </parameter>

      <parameter name="group_2" type="text" groupName="association" multiple="true">
        <label>2: Basic Set</label>
        <description>Basic Set, Control:key1, On/Off control (Button1)</description>
        <multipleLimit>4</multipleLimit>
      </parameter>

    </config-description>

  </thing-type>

</thing:thing-descriptions>

cdjackson, 2016-12-01:
Hmmm - I'm not sure it's a good idea to just edit a different devices XML. We really need to work out why OH isn't creating the XML as even if we add this to the database, it almost certainly won't work - if the XML isn't generating now, then it still won't generate one when the database is added.

As said - if there's no XML generated, then we should get a debug log to work out why.

I just noticed that it looks from the filename like you've created a database XML file - just to note that this is not what I'm after...

hmerk, 2016-12-01:
I know and I will try to do this when I return next week. This was just the easiest way for me to have a quick "fix"

hmerk, 2016-12-07:
@cdjackson after doing a complete reset of my Z-Wave network (only 3 nodes so far) the according XML was created.
https://drive.google.com/file/d/0BxdTuQI5IDvwbG5KOUlFcFZQaHc/view?usp=sharing

cdjackson, 2016-12-18:
Database updated to add the extra type/id.

hmerk, 2016-12-22:
Hmm, there might be an issue with my device, It is not communication properly, but from Binding side, everything seems to be good.

cdjackson, 2017-02-01:
So I'll close this - any further problems please open a new issue...

closed 2017-02-01 by cdjackson

hmerk, 2017-02-01:
Sure :-)