openhab/org.openhab.binding.zwave #244: Qubino Flush 1D relay (ZMHNHD1) not all endpoints in zwave database

reyem, 2016-11-30:
Hi Chris,

Another Qubino device with dynamic endpoints. The device main function works ( exposed switch switches the relay ). However, I attached the optional temperature sensor to it. And this exposes another endpoint on the device (endpoint number 3 according to the manual). Attached you see them in the node xml. I also attached the full openhab log and a filtered one for the particular NODE 25.
According to the manual there would be even an additional sensor be possible attached to I2, which would expose a third endpoint ( endpoint number 2 according to the manual). What puzzles me a bit is the fact, that I have now 2 end points in the XML (id 1 and 2) whereas the manual documents my second endpoint as endpoint 3. However, I don't know if the numbering is important at this point... 

* OH2 Version:
* 2.0.0-SNAPSHOT (Build #609 )

* Device name: Qubino Flush 1D relay
* Device version: ZMHNHD1 H1S4P1
* Link to http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/207

[NODE_25.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/623146/NODE_25.log.txt)
[openhab.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/623147/openhab.log.txt)

[node25.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/623149/node25.xml.txt)

Hope this is enough information for a database update.

Thank you a lot and regards
Marco

P.S. Is it ok to report database adaptions as issue, or would you prefer the community forum for this?

cdjackson, 2016-12-01:
You've written ZMHNHD1 twice, so I assume this isn't a typo, but I can't find this device - is the database link correct - so it's really ZMNHND?

Please confirm before we do anything to the database.

> P.S. Is it ok to report database adaptions as issue, or would you prefer the community forum for this?

Yes - that's fine.

reyem, 2016-12-01:
Sorry, this was a typo combined with the bad habit of copy & paste... thus twice the same typo :-)

You are right, it is ZMNHND1. ( the 1 is really on the device sticker and is for the 868,4 MHz version )

cdjackson, 2016-12-01:
I've updated the database and will update hte binding tonight

reyem, 2016-12-02:
Hi Chris,
I updated to the newest snapshot and confirm to get now all my channels exposed and can properly read values from the temperature sensor as well as switching the relay.
Thank you a lot!
Marco

cdjackson, 2016-12-02:
Great - thanks Marco.

closed 2016-12-02 by cdjackson