openhab/org.openhab.binding.zwave #250: Missing new ProductId for Fibaro gen five devices

beowulfe, 2016-12-04:
I just purchased a new gen five Fibaro Flood Sensor (FGFS101) and Button (FGPB101). Turns out, they have a new ProductId:
http://products.z-wavealliance.org/products/1642
http://products.z-wavealliance.org/products/2017

Pull request incoming.

cdjackson, 2016-12-04:
Please don’t just do a PR - it will be overwritten. Please add the update to the database at www.cd-jackson.com.

beowulfe, 2016-12-04:
Oh, okay. Cancelling the PR.

cdjackson, 2016-12-04:
Thanks.

You need to register to edit, and then email me (although if you do it in the near future I’ll keep an eye out for the registration email and will update your access straight away).

beowulfe, 2016-12-04:
Thanks, I submitted the updates to the database and requested review. I'll close this here.

closed 2016-12-04 by beowulfe

cdjackson, 2016-12-04:
Thanks - I’ll update tonight.