openhab/org.openhab.binding.zwave #253: NPE in ZWaveThingHandler

SuperOok, 2016-12-05:
Hi @cdjackson,

I'm using the zwave2 binding on my Raspbian (Raspberry Pi 2) on OH2.

The binding worked fine for me on OH2-b4. Now I upgraded to latest snapshot, which is build 627 (Zwave  2.0.0.201612010742, Habmin 2.0.0.201612010742. Since then I get regularly (every 10 minutes, sometimes, only every 20 minutes, but exactly 10 or 20 minutes between the occurrences) a set of NPEs:

```
23:17:21.499 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 0: Polling aborted due to exception
java.lang.NullPointerException
        at org.openhab.binding.zwave.handler.ZWaveThingHandler$1.run(ZWaveThingHandler.java:360)[183:org.openhab.binding.zwave:2.0.0.201612010742]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
        at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
23:17:31.797 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - NODE 0: Polling aborted due to exception
java.lang.NullPointerException
        at org.openhab.binding.zwave.handler.ZWaveThingHandler$1.run(ZWaveThingHandler.java:360)[183:org.openhab.binding.zwave:2.0.0.201612010742]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
        at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
```

Not sure what the resulting symptoms are: 
Some of my Zwave devices work fine; 
however, two (Node 6 and 8) stay as 
"Unknown Device" and seem not to get initialized since 2 days now. 
One is a battery sensor, one a mains powered switch actuator.

This is some debug log covering the two

```
23:28:15.947 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:15.999 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
23:28:16.014 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.020 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.034 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.049 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00
23:28:16.051 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 8: Application Command Request (ALIVE:DONE)
23:28:16.053 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 8: Starting initialisation from DONE
23:28:16.055 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@159b0f2 already registered
23:28:16.059 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 8: Incoming command class METER
23:28:16.073 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.076 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 8: Received METER command V3
23:28:16.079 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 8: Meter: Type=Electric(1), Scale=kWh(0), Value=0E+1
23:28:16.085 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveMeterValueEvent
23:28:16.094 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Got an event from Z-Wave network: ZWaveMeterValueEvent
23:28:16.096 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Got a value event from Z-Wave network, endpoint = 0, command class = METER, value = 0E+1
23:28:16.097 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 6: Transaction not completed: node address inconsistent.  lastSent=6, incoming=255
23:28:16.109 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=1
23:28:16.119 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.129 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 14 00 04 00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00 A2
23:28:16.150 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 08 0E 32 02 21 44 00 00 00 00 0E 10 00 00 00 00
23:28:16.158 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 8: Application Command Request (ALIVE:DONE)
23:28:16.162 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 8: Starting initialisation from DONE
23:28:16.168 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@159b0f2 already registered
23:28:16.173 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 8: Incoming command class METER
23:28:16.178 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 8: Received METER command V3
23:28:16.180 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 8: Meter: Type=Electric(1), Scale=kWh(0), Value=0E+1
23:28:16.191 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveMeterValueEvent
23:28:16.193 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Got an event from Z-Wave network: ZWaveMeterValueEvent
23:28:16.195 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Got a value event from Z-Wave network, endpoint = 0, command class = METER, value = 0E+1
23:28:16.198 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 6: Transaction not completed: node address inconsistent.  lastSent=6, incoming=255
23:28:16.201 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
[... 570 lines cut ...]
23:52:46.016 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
23:52:46.018 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8
23:52:46.019 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8
23:52:46.020 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Response[0x01], priority=High, dest=255, callback=0, payload=01
23:52:46.021 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 6: Sent Data successfully placed on stack.
23:52:46.046 [INFO ] [smarthome.event.ItemCommandEvent    ] - Item 'Aussen_Helligkeit' received command 0.0
23:52:46.058 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 43 00 00 05 AD
23:52:46.062 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
23:52:46.064 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 43 00 00 05 00 00 A3
23:52:46.065 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 13 43 00 00 05 00 00 A3
23:52:46.067 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=43 00 00 05
23:52:46.068 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 6: SendData Request. CallBack ID = 67, Status = Transmission complete and ACK received(0)
[... cut ...]
23:52:46.074 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
23:52:46.075 [DEBUG] [commandclass.ZWaveWakeUpCommandClass] - NODE 6: Went to sleep
23:52:46.076 [DEBUG] [commandclass.ZWaveWakeUpCommandClass] - NODE 6: Is sleeping
23:52:46.078 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
23:52:46.079 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 6: Response processed after 76ms/2291ms.
```

Any Idea what could be wrong with Nodes 6 and 8? What's that NPE about?

SuperOok, 2016-12-05:
moved from https://github.com/openhab/openhab2-addons/issues/1526 
Sorry, not yet so accustomed to the different respository structures in the ESH ecosystem

cdjackson, 2016-12-10:
This is probably an issue where devices haven't updated correctly - the node 0 shouldn't happen - it should be filtered out as it means the device hasn't initialised. The errors are related to polling so there won't be any other problem caused by this - just the uninitialised devices won't be updated - which is fine, since they are uninitialised...

So - your real problem is to work out why these two devices are uninitialised.

- What are nodes 6 and 8?
- Can you provide the XML for these devices (while they say unknown, the logs indicates they are actually initialised)?

cdjackson, 2017-01-31:
Closing due to lack of activity. Please feel free to reopen if there's progress.

closed 2017-01-31 by cdjackson