openhab/org.openhab.binding.zwave #255: ZWave send Alarm frame to Fibaro 212 Dimmer manually via rule or openhab switch

apache-123, 2016-12-08:
Hi,

I installed Openhab 2 on windows 10 with 2.0.0 snapshot build 631. I have 25 Fibaro 212 dimmers and would like to be able to manually trigger an alarm behaviour on the dimmers by making the lights flash when i turn on a switch on openhab or triggering them via a rule. The dimmer modules are all on firmware 3.3.  I prefer to use Rule files and Item files etc so a syntax of how to do it would be very much appreciated.

Cheers,
Sathya

cdjackson, 2016-12-08:
It's unclear what you're actually asking for, but as I said in the other thread, for questions, please use the forum. If we find a bug or enhancement in the software then we can open an issue, but I gather this is probably a question.

closed 2016-12-08 by cdjackson

cdjackson, 2016-12-08:
I should add, that if you want to mark the messages on the forum with @chris, then it will hopefully ensure that I see it :)

apache-123, 2016-12-08:
Thanks mate. I have just done that.